tio/tio #205: CYGWIN:  Port disconnection is not detected

HiFiPhile, 2023-09-10:
Disconnect event is not detected on Windows, leaving tio waiting forever. Only option is ctrl+t q exit and relaunch.

I haven't done detailed debug, but I saw 2 threads hang on `WaitForMultipleObjects()` WinAPI, which makes me doubting about Cygwin's `select` implementation when fd is no longer valid.
https://github.com/tio/tio/blob/8134cd3486f939c8520294e952f07a473e1048b6/src/tty.c#L1347C21-L1347C21

lundmar, 2023-09-10:
Right, I did split tio into two threads, one for input and one for output, specifically to avoid tio from becoming unresponsive in case it stalls in its output write call which some have reported. However, I see some Mac users now reporting that they get a select exit failure whenever they quit tio and now this so I may just revert to the single threaded implementation but with timeout handling on i/o.

HiFiPhile, 2023-09-15:
I did more tests, seems like Cygwin just can't detect fd became invalid, following check returns true even if unplugged, also `select()` returns 0 with timeout added.
```
int fd_is_valid(int fd)
{
    return fcntl(fd, F_GETFD) != -1 || errno != EBADF;
}
```

So it seems like either we have to leave it as a known issue, or to add a platform specific serial port backend, some thing like libserialport.

lundmar, 2023-09-16:
Hmm, I wonder what has changed because tio has been available for Cygwin for a long time and no issues reported and I'm sure it worked when we added tio to Cygwin. This is either a regression in tio or Cygwin.

HiFiPhile, 2023-09-18:
Must be Cygwin side, issue is reproducible with this minimal code.

A partial workaround could be watching if the device always exist and Cygwin doesn't have inotify.

```
#include <stdio.h>
#include <unistd.h>
#include <fcntl.h>
#include <sys/ioctl.h>

static int fd;
fd_set rdfs;

int main(int argc, char ** argv) 
{
    int    status;
    struct termios tio;
    char input_char;
    struct timeval tv;

    fd = open(argv[1], O_RDWR | O_NOCTTY | O_NDELAY);
    if (fd == -1) {
        printf("open_port: Unable to open %s ", argv[1]);
        return 0;
    }
    
    memset(&tio, 0, sizeof(tio));
    
    tio.c_cflag |= CS8;
    tio.c_cflag |= CLOCAL | CREAD;
    
    cfsetispeed(&tio, B115200);
    cfsetospeed(&tio, B115200);
    
    tcsetattr(fd, TCSANOW, &tio);
    
    while (1) {

        FD_ZERO( &rdfs);
        FD_SET(fd, & rdfs);

        tv.tv_sec = 1;
        tv.tv_usec = 0;

        status = select(fd + 1, &rdfs, NULL, NULL, &tv);
        if (status < 0)
            perror("select failed");
        else if (status == 0) {
            puts("TIMEOUT");
        } else {

            if (FD_ISSET(fd, & rdfs)) {

                status = read(fd, &input_char, 1);
                printf("Read:%c status:%d\n", input_char, status);
            }
        }
    }

    close(fd);

}
```

lundmar, 2023-09-18:
This must be a regression in Cygwin or maybe the issue is device specific. I assume you are testing with some usb-serial cable or maybe embedded device with on chip usb-serial?

lundmar, 2023-09-18:
I just remember now. I have actually never tested tio on cygwin but on msys2 on Windows running in a VM with a usb-serial device passed through to the VM.

I just tested again with msys2 with the latest tio and at least with msys2 the select call reacts and returns a value when I pull out the FTDI usb-serial device. However, it does return -1 and exits so behavior is slightly different from normal.

[image]

HiFiPhile, 2023-09-18:
> maybe the issue is device specific.

You are right, it behaves differently. I thought it's abstracted by OS so didn't test them all.

Previous tests are done with J-Link's COM port which uses SEGGER's proprietary driver.

|Device|Win Driver|Linux Driver|Disconnect behavior|
|-|-|-|-|
|J-Link|Segger|cdc_acm ttyACM|Undetected |
|CP2102|Silabs|cp210x ttyUSB|  select() failed (Permission denied)|
|CDC class with LPC55 MCU|Windows|cdc_acm ttyACM|  select() failed (Permission denied)|
|SAMA5D2 SAM-BA|Atmel|cdc_acm ttyACM|  select() failed (Permission denied)|

lundmar, 2023-09-18:
Interesting, so something is missing on driver level.

HiFiPhile, 2023-09-18:
Yeah, seems J-Link is a little broken. So good news it's easy to fix generic devices.

HiFiPhile, 2023-09-22:
Closed by #211.

For J-Link workaround is using Window's default CDC driver.

closed 2023-09-22 by HiFiPhile