tio/tio #209: tio MacOSX (Sonoma) file descriptor

k6ef, 2023-09-15:
When I run this: 

~ ➜ tio -b 9600 /dev/tty.usbserial-056700034FFA      

I see this.

[10:11:45.896] tio v2.6
[10:11:45.897] Press ctrl-t q to quit
[10:11:45.904] Connected
[10:11:54.655] Disconnected
[10:11:54.655] Error: select() failed (Bad file descriptor)

I do not see the data I expect from the device. I have tried as my user and as root. I've tried both using the brew install as well as building from source.. 

I've also verified the port/speed settings with other applications.

lundmar, 2023-09-16:
Hmm. So connecting to the same serial device at 9600 with other terminal programs work just fine?

Sonoma is the very latest MacOS version. I wonder if it may have the same regression with select() in the same way as is now being reported in #205 for Cygwin.

However, I think it is most likely a regression in tio. It may be a consequence of being reworked into a two threaded application.

Can I ask you to try manually compile and try tio v2.3? (version before becoming multi threaded).

Thanks.

bjchapm, 2024-01-09:
I see this same error in an earlier Mac configuration -- Ventura 13.6.3.

Running tio 2.7, installed via brew, I get this:

```
$ tio /dev/tty.usbserial-AB7ISYL6
[10:24:49.539] tio v2.7
[10:24:49.539] Press ctrl-t q to quit
[10:24:49.546] Connected
[10:24:56.276] Disconnected
[10:24:56.276] Error: select() failed (Bad file descriptor)
```
Running tio 2.3 (just compiled from source), I see this:
```
./tio /dev/tty.usbserial-AB7ISYL6
[10:26:11.136] tio v2.3
[10:26:11.137] Press ctrl-t q to quit
[10:26:11.143] Connected
[10:26:17.279] Disconnected
```
The error doesn't seem to impact performance and tio seems to work well. Thanks for writing it!

sly74fr, 2024-01-09:
Hi,
Will try it under Ventura tomorrow…

sly74fr, 2024-01-10:
Got the same output on my setup

```
$ tio /dev/tty.usbserial-14430
[15:43:51.363] tio v2.7
[15:43:51.363] Press ctrl-t q to quit
[15:43:51.371] Connected
SDDSSDsdsdssdsd
[15:44:06.184] Disconnected
[15:44:06.184] Error: select() failed (Bad file descriptor)

```

sly74fr, 2024-01-11:
It appears that the `close(fd)` happens once before the last `select(fd,...)` call in `tty_connect()` infinite loop, thus causing the error. Is this working nicely under Linux ?

lundmar, 2024-01-11:
I can't reproduce this issue on Linux. As far as I can tell the code closes the fd handle when select fails the first time (for example when unplugging usb device) and tty_connect will be called again to reconnect and that opens a new fd that should be ok. The terminal stdin input that comes via pipefd[0] from the input thread should be ok too as it is never closed.

Does everything continue to work fine despite the error?

bjchapm, 2024-01-11:
The error does not affect functionality for me on macOS Ventura and everything works fine.

lundmar, 2024-01-11:
Hmm, tio should exit after the error message is printed.

bjchapm, 2024-01-11:
Sorry if my comment wasn't clear. Yes, this message only appears on exit and seems harmless. Otherwise, tio performs correctly. Thanks

lundmar, 2024-01-11:
Ok, I see what is going on here. Apparently select() behaves differently on different platforms.

On Linux, when waiting for input events in the select state, the serial deice descriptor remains valid even when the device is unplugged but it will report errors when read/written to and that is the normal mechanism that tio relies on to reconnect. On mac, it seems the mac kernel invalidates the file descriptor and then select() reports it bad and tio exits. I think this select behavior actually makes more sense but it is what it is.

We have a similar situation on CYGWIN where select() reports EACCES when device is unplugged. On mac select() reports EBADF so we can simply make similar type of special handling for mac.

lundmar, 2024-01-11:
I've made a fix in e572255fd2cac3b0c3bbade9551b201b28c39ca6 . Feel free to test.

sly74fr, 2024-01-11:
Great find !
Will test it tomorrow and report here…

bjchapm, 2024-01-12:
The fix resolves the issue on my mac. Thanks!

closed 2024-02-19 by lundmar