tio/tio #210: What does "play nice with tmux" mean?

JohnAZoidberg, 2023-09-19:
Could you provide some more documentation about that, please? :)

lundmar, 2023-09-19:
Well, it is just a fun remark to stress the point that tio was created as an alternative to screen because it seemed silly to start screen in a tmux session just for serial access.

That being said, tio does not integrate specifically with tmux but it does not get in its way either. For example, tio's default prefix key is ctrl-t while tmux is ctrl-b so no conflict on the prefix key. Also, tio does not use a text UI such as ncurses which means it relies on whatever scroll buffer your terminal provides, including tmux's scroll buffer. In short, tio plays nicely with tmux.

closed 2023-09-21 by lundmar

Zorbatron, 2023-10-11:
One area where tio works with tmux better than minicom is that when you resize the pane in tmux, minicom's history would get cleared.

Zhaotianzheng, 2024-06-01:
When I use tmux copy mode with tio, the scrollback buffer will change when there is new lines in tio and cause a hard time to select something in scrollback buffer. Is there some possible solution?

Eg. 
tio have line below, and i want to select aaa
aaa
bbb
ccc
And when I select the output becomes
aaa
bbb
ccc
ddd
My selection will become select bbb, it is quite a bummer...

lundmar, 2024-06-01:
This is not really a tio issue but a tmux issue.

Copy/paste has always been a bit painful in tmux so it may be easier for you to simply not use tmux in situations where you need to copy/paste a lot.

That being said I don't see the problem you are describing because normally when tmux enters copy mode the pane text should freeze and ignore any further input until you have selected and copied your text so you should not be disturbed by incoming lines.

This should be the same behavior for your tmux unless maybe you are using a very old tmux version.

Here are the tmux copy/paste steps:
```
ctrl+b [ (enter copy mode)
ctrl+space (start selection)
ctrl+w (copy)
ctrl+b ]  (paste)
```

Zhaotianzheng, 2024-06-02:
@lundmar Thanks. It turns out the problem of my tmux, after reinstall, it works fine.