tio/tio #212: ctrl+t not opening command menu

timonsku, 2023-10-21:
I'm connected to a Linux machine via SSH from Windows 10 using Windows Terminal and built-in SSH.
Unfortunately it seems like Ctrl+T is possibly swallowed or for a different reason not registered but I am not able to exit a session via ctrl+T Q. Not sure if its because of the SSH session or not though.

I did work briefly with a different device that was logging less data but I had to smash the combination a bunch of times.
What I'm seeing on the device where this is never working is that tio stops logging the time stamps so maybe this is due to the way the device is sending data and tio getting confused about the stream?

[image]

lundmar, 2023-10-21:
Hmm, that does sound strange.

Not being able to exit, does that happen immediately after connecting or after a while?

What happens if you log in to the system but with no ADC sampling running?

timonsku, 2023-10-21:
immediately after connecting, not printing any values seems to cause the same issue so probably nothing to do with the data rate.
It works in a different firmware built with the same framework, a bit puzzling.
If you have an ESP32 board available I could provide binaries.

timonsku, 2023-10-21:
Or if you feel like building Zephyr: https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/drivers/adc
Have not tested if this happens too if its build for a different board.

lundmar, 2023-10-21:
I'm using Zephyr on the daily - I've never seen this behavior.

What Linux distro are you using?

timonsku, 2023-10-21:
The remote machine runs Ubuntu 22.04

It does not happen in my own firmware flashed to the same device weirdly enough. Something specific to that ADC sample.

lundmar, 2023-10-21:
It may be that somehow the firmware is spewing out some control characters that messes with not tio but the terminal that you run tio in, for example gnome-terminal.

Whatever terminal you are using, could you try another terminal such as xterm?

timonsku, 2023-10-21:
it's run over SSH, Windows Terminal is what I'm using: https://github.com/microsoft/terminal
Same thing with Putty.
On the machine itself quitting seems to work but it does not show the little green command input that tio usually shows either but it does listen to the input at least.

In theory it should not print anything at all, I removed all kernel logs. After the `Connected` message there is nothing printed (visibly)

lundmar, 2023-10-21:
> it's run over SSH, Windows Terminal is what I'm using: https://github.com/microsoft/terminal Same thing with Putty. On the machine itself quitting seems to work but it does not show the little green command input that tio usually shows either but it does listen to the input at least.
>

So you are saying that tio works normally if you do not connect via ssh?

I'm not sure what you mean by the little green command input?

timonsku, 2023-10-21:
it accepts input but behaves still strange.
Normally I would expect to see this:
[image]

that `uart:~$` input field that stays always at the bottom never appears in either of the 3 terminal sessions but works fine in a different firmware (same target, sdk etc) in all 3 terminals (ssh or not)

lundmar, 2023-10-22:
The green prompt and timestamps not showing seems to be a difference in firmware configuration. If you notice, the timestamps in your screenshot is actually not tio's timestamp format (tio timestamps do not do the ',000' thing. Meaning, it your firmware that is built with/without timestamping. I figure the same goes for the prompt feature.

timonsku, 2023-10-22:
oh you're right! I've only been using tio (since yesterday) with Zephyr so far :)

Well the issue with that one specific sample is nothing crucial for me. Just thought I'd report it.
Unclear to me though what exactly the hang up is but likely todo with the interactive terminal being active in my firmware.

lundmar, 2023-10-22:
So the only problem here is the inability to quit tio or have it respond to commands?

Perhaps you are not pressing the prefix key sequence correctly?

Pres ctrl, press t, release both, press q ?

timonsku, 2023-10-22:
Yes, the swallowed input is the only issue then and only happens over SSH, not when using a terminal on the machine itself.
Am positive that I press the sequence correctly, like you describe and it works immediately when I use it in the firmware with interactive shell enabled.

lundmar, 2023-10-22:
Hmm, I just tried run up Windows 10 in a VM guest and then use ssh in a Power Shell to connect to my Ubuntu VM host and from there use tio. I have no problem quitting or sending other tio commands.

It must be a problem specific to your ssh setup.

timonsku, 2023-10-22:
Are you running that specific sample on an ESP32 or something else?
Cause I only have the issue in that specific scenario. Everything else works just fine so far.

lundmar, 2023-10-22:
No, I don't have any ESP hardware. I'm testing with a STM32 micro that I'm currently writing a zephyr driver for.

It can't relate to the firmware as tio intercepts any input before it is passed on to the target so something like quit should just work.

It must be something specific to your ssh configuration so that maybe the ssh client intercepts your e.g. ctrl strokes or something.

timonsku, 2023-10-22:
I'm not sure, on one hand yes, certainly somewhere in that chain but if it was very inherent then I should see this with other firmware over this SSH setup as well?

Oddly enough after trying again it now seems to be okay? Very weird, it did not for the past dozen or so tests with that sample without changing anything in the setup or even session.
But it certainly seems to be a fringe thing. I can keep an eye on it if I see it appear in other situations again but feel free to close for now.

lundmar, 2023-10-22:
No matter what firmware you have installed on your target, you should be able send commands to to tio. All tio input processing including handling tio commands is done before tio passes input data to the target.

Feel free to reopen the issue if you can reproduce the issue.

closed 2023-10-22 by lundmar