tio/tio #215: ModuleNotFoundError: No module named 'mesonbuild'

keelung-yang, 2023-12-07:
Hi,
I'm trying to install tio by compiling on Ubuntu 20.04.6:

Install cmake
```bash
sudo apt purge --auto-remove cmake
wget -O - https://apt.kitware.com/keys/kitware-archive-latest.asc 2>/dev/null | gpg --dearmor - | sudo tee /etc/apt/trusted.gpg.d/kitware.gpg >/dev/null
sudo apt-add-repository 'deb https://apt.kitware.com/ubuntu/ focal main'     
sudo apt update
sudo apt install cmake scdoc libinih-dev
```

```bash
git clone https://github.com/tio/tio.git
cd tio && meson setup build
The Meson build system
Version: 1.3.0
Source dir: /home/keelung/ssd/work/yang/tio
Build dir: /home/keelung/ssd/work/yang/tio/build
Build type: native build
Project name: tio
Project version: 2.8
C compiler for the host machine: cc (gcc 9.4.0 "cc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0")
C linker for the host machine: cc ld.bfd 2.34
Host machine cpu family: x86_64
Host machine cpu: x86_64
Check usable header "asm-generic/ioctls.h" : YES 
Header "asm-generic/ioctls.h" has symbol "TCGETS2" : YES 
Fetching value of define "B0" : 0000000 
Fetching value of define "B50" : 0000001 
Fetching value of define "B75" : 0000002 
Fetching value of define "B110" : 0000003 
Fetching value of define "B134" : 0000004 
Fetching value of define "B150" : 0000005 
Fetching value of define "B200" : 0000006 
Fetching value of define "B300" : 0000007 
Fetching value of define "B600" : 0000010 
Fetching value of define "B1200" : 0000011 
Fetching value of define "B1800" : 0000012 
Fetching value of define "B2400" : 0000013 
Fetching value of define "B4800" : 0000014 
Fetching value of define "B7200" : (undefined) 
Fetching value of define "B9600" : 0000015 
Fetching value of define "B14400" : (undefined) 
Fetching value of define "B19200" : 0000016 
Fetching value of define "B28800" : (undefined) 
Fetching value of define "B38400" : 0000017 
Fetching value of define "B57600" : 0010001 
Fetching value of define "B76800" : (undefined) 
Fetching value of define "B115200" : 0010002 
Fetching value of define "B230400" : 0010003 
Fetching value of define "B460800" : 0010004 
Fetching value of define "B500000" : 0010005 
Fetching value of define "B576000" : 0010006 
Fetching value of define "B921600" : 0010007 
Fetching value of define "B1000000" : 0010010 
Fetching value of define "B1152000" : 0010011 
Fetching value of define "B1500000" : 0010012 
Fetching value of define "B2000000" : 0010013 
Fetching value of define "B2500000" : 0010014 
Fetching value of define "B3000000" : 0010015 
Fetching value of define "B3500000" : 0010016 
Fetching value of define "B4000000" : 0010017 
Check usable header "linux/serial.h" : YES 
Header "sys/ioctl.h" has symbol "TIOCSRS485" : YES 
Configuring config.h using configuration
Run-time dependency threads found: YES
Found pkg-config: YES (/usr/bin/pkg-config) 0.29.1
Run-time dependency inih found: YES 48
Configuring tio using configuration
Run-time dependency bash-completion found: YES 2.10
Configuring tio.1 using configuration
Build targets in project: 1

Found ninja-1.10.0 at /usr/bin/ninja
```

```bash
keelung@hpz:~/ssd/work/yang/tio$ meson compile -C build
INFO: autodetecting backend as ninja
INFO: calculating backend command to run: /usr/bin/ninja -C /home/keelung/ssd/work/yang/tio/build
ninja: Entering directory `/home/keelung/ssd/work/yang/tio/build'
[16/16] Linking target src/tio
```

```bash
keelung@hpz:~/ssd/work/yang/tio$ meson install -C build
ninja: Entering directory `/home/keelung/ssd/work/yang/tio/build'
ninja: no work to do.
Installing src/tio to /usr/local/bin
Installation failed due to insufficient permissions.
Attempt to use /usr/bin/sudo to gain elevated privileges? [y/n] y
Traceback (most recent call last):
  File "/home/keelung/.local/bin/meson", line 5, in <module>
    from mesonbuild.mesonmain import main
ModuleNotFoundError: No module named 'mesonbuild'
keelung@hpz:~/ssd/work/yang/tio$ 
```

keelung-yang, 2023-12-08:
Thanks! 
--prefix should be work.
As I remembered I use sudo pip3 install .. to finish installing tio
So I guess both tio and meson should be either in local user or system wide.
Finally since it's solved, I close it by myself.

closed 2023-12-08 by keelung-yang

AtaraxiaZ, 2024-05-10:
Another solution is to install meson using 
```
sudo pip3 install meson
```

Then the mesonbuild module could be found, and tio could be installed under `/usr/local/bin`.