tio/tio #218: Impossible to connect device in Windows10

psv84, 2023-12-29:
When I'm trying to connect device in linux everything is working fine.
When I'm trying to do the same on Windows 10 it is just freeze during connection ...
[image]
Perhaps this is the known issue and somebody can help ?
Thanks in advance

psv84, 2023-12-29:
Looks like it is related to chip version HXA that is not supported on Win10 according to https://www.prolific.com.tw/US/ShowProduct.aspx?p_id=225&pcid=41

lundmar, 2023-12-29:
Yeah, this looks like a Windows hardware support issue. Amazing that MS just removes driver support for EOL but popular chip versions like that.

lundmar, 2023-12-30:
Closing issue as this seems to be a hardware issue.

closed 2023-12-30 by lundmar