tio/tio #219: Option to pulse serial line on (re)connect

f0x52, 2024-01-07:
Building off of https://github.com/tio/tio/issues/153 / https://github.com/tio/tio/pull/154, for microcontroller usage it would be quite convenient to have a command line flag that pulses a serial line (DTR in my case) when tio connects. Just to save the extra manual `ctrl-t p 0` every time

f0x52, 2024-01-07:
For anyone interested, I quickly hacked it in (which was very easy, good codebase) https://github.com/tio/tio/compare/master...f0x52:tio:dtr-pulse

But for upstreaming (if desired at all) it'd probably be better to have it configurable for all line types etc, just didn't have the time for that rn.

lundmar, 2024-02-19:
Sorry for the delayed response. I understand how this is useful. However, as you hint yourself, I would like to see such feature become fully configurable. This would also include defining initial pulse state. Right now tio simply toggles the line.

Actually, instead of introducing some potentially cumbersome inflexible command line interface for this feature I am considering introducing a general lua scripting backend in tio that will initially support this feature but which can later on be extended with other useful scripting features to support more complex interaction with the connected device (maybe some expect like functionality for automatic responses etc.).

With a lua script backend tio could run the following lua script at startup:
```
high(DTR); msleep(100); low(DTR); msleep(50); toggle(DTR)
```

Either defined in a separate file or via variable in the tio configuration file.

lundmar, 2024-04-01:
The lua scripting feature to support flexible programming of the control lines upon connect is now implemented in 0becfa3274c80cb8d18fed0325c776eb4bfdb1ad. See #233 for more information.

f0x52, 2024-04-12:
Amazing!