tio/tio #223: Try really hard not to exit on socket SIGPIPE?

GWHAYWOOD, 2024-02-20:
Now that tio has been in use here for around 18 months, first of all, thank you very much for tio. :)

In what I'm going to call 'normal' use we see no problems.  But we use it otherwise, quite a lot.

The issue we see most is that in 'not normal' use, occasionally, tio will exit unexpectedly.

The 'not normal' use is when (under the control of some remote user) a process on the box running tio connects to tio's socket to send and receive the serial port data to and from the remote user.

In this scenario both the user of the Xterm running tio and the remote process are sharing the serial port connection.  That's OK, because the users are in the same office and they arrange with each other to share access to tio and to the serial port.  But it is not OK if tio exits and hangs things up.

The occasion is (mostly, as far as I can tell) when the local listening process which has connected to tio's socket stops listening to it, but tio is still sending to it.  What it's sending might only be some boilerplate which the remote user doesn't need to see.  It's kinder to the user not to make him wait for it all to scroll past at 19.2kbit/s, so we don't make him wait.

If we get that 'do not make him wait' bit wrong, the result is usually an operating system SIGPIPE error, which tio faithfully returns via the shell when it exits.

Unfortunately then of course the remote process then can no longer communicate via tio's socket with the serial device to which tio was connected - because the socket has gone away.

Unfortunately, also, if an Xterm was used to start tio, the Xterm is then left in a broken state and a 'reset' command (probably typed blindfold) is needed to recover the Xterm before tio - or anything else - can be restarted in the Xterm.

It is probably going to be possible to avoid these occasions by more careful coding of the software which uses tio for its communication with the serial device, but that is going to take time to get right.  In the meatime, it would be really nice if there was a way to tell tio *not* to exit under these circumstances.

Is there a way to tell tio - except for when it gets the CTRL-T Q command - to try really hard not to exit but to carry on as best it can?

FWIW I don't think 'nohup' is a solution here.

lundmar, 2024-05-10:
Hi @GWHAYWOOD, sorry for the late response. I've been busy IRL but now I'm back for some tio maintenance and have been working backwards through the issue list.

Can you please paste the exact error message that tio prints when it exits?

Thanks.

GWHAYWOOD, 2024-05-10:
No worries about the delay.

There is no error message.  Tio just quits, and drops the user back to the command line which started it.

lundmar, 2024-05-10:
Hmm. Do you close the tcp socket connection at any time or do you just stop reading from the socket and then after a while tio will quit?

GWHAYWOOD, 2024-05-10:
I've looked at our code (for the first time in quite a while) and I'm pretty sure that Tio quits on the broken pipe signal before we make any attempt to close the pipe.

We do close the pipe on a 'normal' exit, but 'normal' means that we've consumed all the data from the pipe and that seems to cause no problem.  The user (the user who started Tio, and who is using Tio, in parallel with our process, as a terminal for the system being accessed via a serial port connection) is not affected in this case, and can continue using the terminal as they normally would.

Having said that, the coder who actually did this work is away in France for a few days but will be back around the middle of next week.  Then if you want us to try any suggestions we will be able to do some debugging on our bench setup, which is as near to identical with the equipment in production as we can make it.

Once again, thank you for your work on Tio.  It is very much appreciated.

lundmar, 2024-05-10:
> I've looked at our code (for the first time in quite a while) and I'm pretty sure that Tio quits on the broken pipe signal before we make any attempt to close the pipe.

The puzzling part is that tio is supposed to ignore any SIGPIPE signals. This was added in db765bc3.

Since you said tio does not print any error message I assume you have instrumented the code to conclude that a SIGPIPE occurs?
 
> Having said that, the coder who actually did this work is away in France for a few days but will be back around the middle of next week. Then if you want us to try any suggestions we will be able to [...]

In db765bc3 I have change the socket code to explicitly ignore any signals when it tries to write to the socket. I suspect the problem may be elsewhere but give it a try.

> Once again, thank you for your work on Tio. It is very much appreciated.

Thank you. I'm quite happy with the new tio v3.x series and I hope users are too.

GWHAYWOOD, 2024-05-10:
Yes, we instrumented (our own) code to log the SIGPIPE events.  The shell which started Tio also returns the SIGPIPE exit code 141 instead of exit code 0.  We are very confident that this is the issue.  It usually takes a few minutes to get one of these events to happen but it's only a matter of repeating the same operation several times until it happens.  In our setup it happens somewhere between one in five and one in twenty attempts.

We will try f71ffea... next week, thank you.

lundmar, 2024-05-15:
@GWHAYWOOD Great. Thank you for testing.

closed 2024-05-15 by lundmar