tio/tio #226: Is timestamp disabled in hex mode?

sgon00, 2024-02-29:
Hi, I am trying to enable timestamp in hex mode. But there is no timestamp. 
When I read this issue: https://github.com/tio/tio/issues/104
It shows timestamp disabled in hex mode. I don't really understand that.
If I don't use hex mode, the captures are garbled characters. So I have to use hex mode.
But without timestamp, the captures are messy.

sgon00, 2024-03-01:
> In the current hex mode this does not make sense since this is meant to be a raw data dump mode.

I really don't understand why this does NOT make sense in hex mode.
I think it makes much sense to have timestamp on every new line data.
I am capturing the request and response modbus data from RS485 port.
I can show the example outputs I had with `tio` in linux and some tool called `sscom5` in windows:

```
[11:18:07.200] Press ctrl-t q to quit
[11:18:07.209] Connected
01 03 01 80 00 0a c5 d9 01 03 14 00 95 00 97 00 e1 00 e1 00 00 00 c8 03 26 00 00 03 25 03 26 a1 b5 01 03 01 80 00 0a c5 d9 01 03 14 00 95 00 97 00 e1 00 e1 00 00 00 c8 03 26 00 00 03 25 03 26 a1 b5 01 03 01 80 00 0a c5 d9 01 03 14 00 95 00 97 00 e1 00 e1 00 00 00 c8 03 26 00 00 03 25 03 26 a1 b5 01 03 01 80 00 0a c5 d9 01 03 14 00 95 00 97 00 e1 00 e1 00 00 00 c8 03 26 00 00 03 25 03 26 a1 b5 01 03 01 80 00 0a c5 d9 01 03 14 00 95 00 97 00 e1 00 e1 00 00 00 c8 03 26 00 00 03 25 03 26 a1 b5 01 03 01 80 00 0a c5 d9 01 03 14 00 95 00 97 00 e1 00 e1 00 00 00 c8 03 26 00 00 03 25 03 26 a1 b5 01 03 01 80 00 0a c5 d9 01 03 14 00 95 00 97 00 e1 00 e1 00 00 00 c8 03 26 00 00 03 25 03 26 a1 b5 01 03 01 80 00 0a c5 d9 01 03 14 00 95 00 97 00 e1 00 e1 00 00 00 c8 03 26 00 00 03 25 03 26 a1 b5 01 03 01 80 00 0a c5 d9 01 03 14 00 95 00 97 00 e1 00 e1 00 00 00 c8 03 26 00 00 03 25 03 26 a1 b5 01 03 01 80 00 0a c5 d9 01 03 14 00 95 00 97 00 e1 00 e1 00 00 00 c8 03 26 00 00 03 25 03 26 a1 b5 01 03 01 80 00 0a c5 d9 01 03 14 00 95 00 97 00 e1 00 e1 00 00 00 c8 03 26 00 00 03 25 03 26 a1 b5 01 03 01 80 00 0a c5 d9 01 03 14 00 95 00 97 00 e1 00 e1 00 00 00 c8 03 26 00 00 03 25 03 26 a1 b5 01 03 01 80 00 0a c5 d9 01 03 14 00 95 00 97 00 e1 00 e1 00 00 00 c8 03 26 00 00 03 25 03 26 a1 b5 01 03 01 80 00 0a c5 d9 01 03 14 00 95 00 97 00 e1 00 e1 00 00 00 c8 03 26 00 00 03 25 03 26 a1 b5 01 03 01 80 00 0a c5 d9 01 03 14 00 95 00 97 00 e1 00 e1 00 00 00 c8 03 26 00 00 03 25 03 26 a1 b5 01 03 01 80 00 0a c5 d9 01 03 14 00 95 00 97 00 e1 00 e1 00 00 00 c8 03 26 00 00 03 25 03 26 a1 b5 
```

The above output is a very easy and clean one, because the request data never changes and the response data never changes too. But it's already very messy to identify the frequency of request/response data and what they are.

The following output is what I have in `sscom5` on windows:

```
[09:39:01.456]←◆01 03 01 80 00 0A C5 D9 01 03 14 00 95 00 97 00 E1 00 E1 00 00 00 C8 03 24 00 00 03 23 03 24 E3 B5 
[09:39:03.456]←◆01 03 01 80 00 0A C5 D9 01 03 14 00 95 00 97 00 E1 00 E1 00 00 00 C8 03 24 00 00 03 23 03 24 E3 B5 
[09:39:05.459]←◆01 03 01 80 00 0A C5 D9 01 03 14 00 95 00 97 00 E1 00 E1 00 00 00 C8 03 24 00 00 03 23 03 24 E3 B5 
[09:39:07.456]←◆01 03 01 80 00 0A C5 D9 01 03 14 00 95 00 97 00 E1 00 E1 00 00 00 C8 03 24 00 00 03 23 03 24 E3 B5 
[09:39:09.456]←◆01 03 01 80 00 0A C5 D9 01 03 14 00 95 00 97 00 E1 00 E1 00 00 00 C8 03 24 00 00 03 23 03 24 E3 B5 
[09:39:11.456]←◆01 03 01 80 00 0A C5 D9 01 03 14 00 95 00 97 00 E1 00 E1 00 00 00 C8 03 23 00 00 03 23 03 24 95 75 
[09:39:13.458]←◆01 03 01 80 00 0A C5 D9 01 03 14 00 95 00 97 00 E1 00 E1 00 00 00 C8 03 23 00 00 03 23 03 24 95 75 
[09:39:15.457]←◆01 03 01 80 00 0A C5 D9 01 03 14 00 95 00 97 00 E1 00 E1 00 00 00 C8 03 23 00 00 03 23 03 24 95 75 
[09:39:17.459]←◆01 03 01 80 00 0A C5 D9 01 03 14 00 95 00 97 00 E1 00 E1 00 00 00 C8 03 23 00 00 03 23 03 24 95 75 
[09:39:19.456]←◆01 03 01 80 00 0A C5 D9 01 03 14 00 95 00 97 00 E1 00 E1 00 00 00 C8 03 23 00 00 03 23 03 24 95 75 
[09:39:21.459]←◆01 03 01 80 00 0A C5 D9 01 03 14 00 95 00 97 00 E1 00 E1 00 00 00 C8 03 23 00 00 03 23 03 24 95 75 
[09:39:23.456]←◆01 03 01 80 00 0A C5 D9 01 03 14 00 95 00 97 00 E1 00 E1 00 00 00 C8 03 23 00 00 03 23 03 24 95 75 
[09:39:25.456]←◆01 03 01 80 00 0A C5 D9 01 03 14 00 95 00 97 00 E1 00 E1 00 00 00 C8 03 23 00 00 03 23 03 24 95 75 
[09:39:27.457]←◆01 03 01 80 00 0A C5 D9 01 03 14 00 95 00 97 00 E1 00 E1 00 00 00 C8 03 24 00 00 03 23 03 24 E3 B5 
[09:39:29.459]←◆01 03 01 80 00 0A C5 D9 01 03 14 00 95 00 97 00 E1 00 E1 00 00 00 C8 03 24 00 00 03 23 03 24 E3 B5 
```

`sscom5` output is very clean and easy to view. The request/response sent every two seconds.
`01 03 01 80 00 0A C5 D9` is the request data.
`01 03 14 00 95 00 97 00 E1 00 E1 00 00 00 C8 03 24 00 00 03 23 03 24 E3 B5` is the response data.
`01` is the slave id. `03` is the function code. `00 0A` is the number of requested registers. Depending on this number, the response length (number of bytes) will be changed. So the number of bytes in reponse are not fixed. I will not explain what the following hex codes mean here unless you are interested in.
They are just typical modbus data. They are meaningful in hex format and can not be viewed in ASCII mode.
I really love linux, but so far, I can not find any tools to support timestamp in hex mode unless I am writing one later if I have time.
I don't want to use windows, but this is the life, I have to live with it.

lundmar, 2024-03-01:
> > In the current hex mode this does not make sense since this is meant to be a raw data dump mode.
> 
> I really don't understand why this does NOT make sense in hex mode. I think it makes much sense to have timestamp on every new line data. 

You misunderstand. In normal non-hex mode we literally listen for the new live character (0x0a) to arrive and when it arrives we print a new time stamp.

What I meant is that this same mechanism can not be reused in hex mode since it will then print a new timestamp whenever you encounter arbitrary data containing the newline value. Your modbus request header is an example of exactly that. This is why we disabled new line support in the current hex dump implementation.

> The following output is what I have in `sscom5` on windows:
>
> ```
> [09:39:01.456]←◆01 03 01 80 00 0A C5 D9 01 03 14 00 95 00 97 00 E1 00 E1 00 00 00 C8 03 24 00 00 03 23 03 24 E3 B5 [...]

No need to explain further. I'm very familiar with modbus since I have integrated RS-485 openmodbus on several micro-controllers.

It is clear to me that sscom5 must be using a timer based timestamp as it is clearly not printing new time stamps based on the new line character. Also, for this feature to be useful there must be some sort of configurable timestamp timeout value available so you can tune it to your specific use case. I have not seen such feature in any hex dumper before - some would probably consider it a bit of an exotic feature. However, I can see the usefulness of it so I have added the feature to our [TODO](https://raw.githubusercontent.com/tio/tio/master/TODO) list.

> I really love linux, but so far, I can not find any tools to support timestamp in hex mode unless I am writing one later if I have time. I don't want to use windows, but this is the life, I have to live with it.

Nobody wants to use Windows for development. That is just a silly thought! 😁

Since you love Linux I expect you to have a better understanding of how open source works. It seems quite rude of you to quickly and publicly state that an entire open source project is USELESS simply because it does not currently support some exotic feature that you impatiently feel that you need right now.

[image]

You stated this on [stackoverflow.com](https://stackoverflow.com/questions/20541519/is-there-a-linux-serial-terminal-with-timestamps/74914775?noredirect=1#comment137647925_74914775).

This is not the best way to help motivate open source developers to add support for your requested feature. Just saying.

lundmar, 2024-03-01:
@sgon00 Apology accepted.

The feature is on the TODO list. I expect that the traditional hex output format will be implemented first. After that we will have to look into adding the timer based timestamping.

sgon00, 2024-03-01:
@lundmar I saw the TODO list. Thank you very much for your explanation and time.

closed 2024-04-18 by lundmar