tio/tio #228: Visual bell?

GWHAYWOOD, 2024-03-06:
Many terminals and terminal emulators implement a visual bell.

I've scoured the man page for any hint, and grepped the tio source for everything I can think of, but nothing is telling me that tio has such a feature.

I'd find it useful.  Anyone else?

lundmar, 2024-03-06:
From the man page:

```
       --alert none|bell|blink

              Set alert action on connect/disconnect.

              It will sound the bell once or blink once on successful connect. Likewise it will sound the bell twice or blink twice on disconnect.

              Default value is "none".
```

Blink is your visual bell.

GWHAYWOOD, 2024-03-06:
Thanks for the quick reply, and I'm sorry I was not clear.

I meant in response to the ASCII character '0x07' (CTRL-G) arriving from the serial port.

lundmar, 2024-03-06:
Actually, that has nothing to do with tio - tio is not a terminal application. Tio simply handles input characters from your serial device and prints them to your terminal application and vice versa. Meaning, if you receive the visual bell character, then whatever terminal application (gnome-terminal, xterm, etc.) that you are using should react to it and do the visual blink.

lundmar, 2024-03-08:
Closing issue. Feel free to reopen if you need more information.

closed 2024-03-08 by lundmar