tio/tio #229: Serial connection established but dont see any logs

anushasrivats, 2024-03-15:
hi this is my first time experimenting with serial logs. I have established connection using :
- tio --list-devices : to get the USB->serial device
-  sudo tio /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_BMADb133812-if00-port0 -l --log-file /tmp/ttyusb0.txt : to connect to the desired device.
- While I get : [10:28:08.206] tio v2.8
[10:28:08.206] Press ctrl-t q to quit
[10:28:08.207] Connected
 and I am able to disconnect and connect back my device,  i see no logs in /tmp/ttyusb0.txt

lundmar, 2024-03-15:
I just tried run the same command:
```
lundmar@wopr:~/projects/tio/tio$ sudo /home/lundmar/opt/tio/bin/tio /dev/serial/by-id/usb-STMicroelectronics_STLINK-V3_004900343438510234313939-if02 -l --log-file /tmp/ttyusb0.txt
[20:12:15.781] tio v2.8
[20:12:15.781] Press ctrl-t q to quit
[20:12:15.782] Connected
[20:12:21.076] Disconnected
[20:12:22.077] Warning: Could not open tty device (No such file or directory)
[20:12:22.078] Waiting for tty device..
[20:12:25.082] Connected
[20:12:30.642] Disconnected
[20:12:30.642] Saved log to file /tmp/ttyusb0.txt
```
This produces the log mentioned. Even if I try connect a non-existing serial device it will still create a log file.

What platform are you testing on?

How did you install tio?

anushasrivats, 2024-03-15:
I installed from sources, following the README document. Yes it creates a log file. but it is always empty. I am running on a BeaglePlay board and want the boot logs. I have tried connecting the device and booting the board multiple times but the log file is always empty.

lundmar, 2024-03-15:
If you are seeing an empty log file it may be because you are not receiving any characters. Do you see any activity on your serial device?

anushasrivats, 2024-03-15:
how to confirm that?

lundmar, 2024-03-15:
Simply watch the output in tio.

anushasrivats, 2024-03-15:
no output. It say Connected and then there is no activity. no output, nothing in log file. 
if after connecting, i am booting the board, sholdnt tio show boot logs though?

lundmar, 2024-03-15:
Sounds like you are not connecting to the correct serial device since you do not see any output during boot of your board. This is why tio is logging nothing to the log file.

Make sure that you are connecting to the correct serial port and with the correct speed, parity, etc. configuration.

anushasrivats, 2024-03-15:
i had interchanged TX and RX cables. its working now :) thanks!

anushasrivats, 2024-03-15:
had interchanged TX and RX. serial connection working now. closing issue

closed 2024-03-15 by anushasrivats

lundmar, 2024-03-15:
A classic mistake! 😀