tio/tio #230: Mingw: doesn't rely on system line state

HiFiPhile, 2024-03-18:
There is a Mingw issue which doesn't read line states correctly depending the converter model, reported states always stays the same while state change actually works:
```
[17:18:28.292] Line states:
[17:18:28.292]  DTR: HIGH
[17:18:28.292]  RTS: HIGH
[17:18:28.292]  CTS: HIGH
[17:18:28.293]  DSR: LOW
[17:18:28.293]  DCD: LOW
[17:18:28.293]  RI : LOW
```

- Affected: Silabs CP2102N; WCH CH9102F
- Doesn't affected: ttyACM; FTDI FT232RQ

Due to this bug line toggle and line pulse doesn't work as they read states from the system:
```
[17:25:12.133] Please enter which serial line number to toggle:
[17:25:12.133]  DTR (0)
[17:25:12.133]  RTS (1)
[17:25:12.133]  CTS (2)
[17:25:12.133]  DSR (3)
[17:25:12.134]  DCD (4)
[17:25:12.134]  RI  (5)
[17:25:12.531] Setting DTR to LOW
[17:25:14.211] Please enter which serial line number to toggle:
[17:25:14.211]  DTR (0)
[17:25:14.211]  RTS (1)
[17:25:14.211]  CTS (2)
[17:25:14.211]  DSR (3)
[17:25:14.211]  DCD (4)
[17:25:14.211]  RI  (5)
[17:25:14.452] Setting DTR to LOW
```
```
[17:27:49.004] Please enter which serial line number to pulse:
[17:27:49.005]  DTR (0)
[17:27:49.005]  RTS (1)
[17:27:49.005]  CTS (2)
[17:27:49.005]  DSR (3)
[17:27:49.005]  DCD (4)
[17:27:49.005]  RI  (5)
[17:27:49.359] Setting DTR to LOW
[17:27:49.361] Waiting 100 ms
[17:27:49.473] Setting DTR to LOW
```

I think it's better to remember line states inside application.

lundmar, 2024-03-18:
This looks like a HW driver issue to me - apparently the driver fails to read out the line state.

It does not help much to manage the line state internally if we can't trust the read out of line state in the first place.

tio is doing the right thing by always checking the line state at init time and line toggle.

HiFiPhile, 2024-03-18:
> This looks like a HW driver issue to me

It's more funky than I thought, on two PC with the same updated MingW version, same CP2102 dongle, same tio v2.7 and same driver, one line state works but others not...

I'm closing and will do more investigation...

closed 2024-03-18 by HiFiPhile

lundmar, 2024-03-18:
Good luck. Maybe check if they are running same Windows version and same driver versions.