tio/tio #231: switching between mark and space parity while connection is established

marvinhinz, 2024-03-20:
Hi!

we currently trying to talk to a weird rs-485 multi drop bus controller via ttyusb. The protocol requires 9 bits, the 9th bit declares if the current byte is a node address or not. For our implementation this would require to switch between mark and parity while sending single bytes. Also to reset the device, we would need to hold the DTS and RTS linus up for 100ms. 

Is that somehow supported within tio? I couldnt really tell from the documentation.

Thanks!

lundmar, 2024-03-20:
Yes, once connected you can toggle or pulse the control lines.

However you can not reconfigure any of the port settings such as paritys during a session.

marvinhinz, 2024-03-20:
> Yes, once connected you can toggle or pulse the control lines.
> 
> However you can not reconfigure any of the port settings such as parity during a session.

Thanks for the swift response, is this a technical limitation or would it be feasible to add an api for that? Sadly im not very fluent in c, but i would try to submit a pr if its generally possible.

lundmar, 2024-03-20:
No, there is no technical limitation. There is an options structure that you can manipulate and then call tty_configure() which will configure a new terminal structure (tio) from the options which can then be used with tcsetattr() to apply the new serial port configuration. You will understand when you read the code.

The hardest part is possibly handling the input commands and make sure they make sense naming and input wise.

Today ctrl-t c shows the configuration.

If you do go ahead with this feature I suggest to add a ctrl-t C command to do any port reconfiguration. Also, the feature should be able to do general port settings.

Meaning, ctrl-t C d 7 could change data bits to 7.
Likewise, ctrl-t C s 2 could change stop bits to 2 etc.

For inspiration see how ctrl-t p works.

closed 2024-04-10 by lundmar