tio/tio #233: Enhance line pulse function

HiFiPhile, 2024-03-29:
Espressif ESP32 boards and some Zigbee devices like SonOff ZBDongle-P use a special RTS/DTR sequence to reset MCU or enter bootloader.
In the schematic `EN` is MCU reset and depends on `IO0` state when `EN` become 1, MCU will enter user application or bootloader:
[image]

### To enter bootloader:
1. DTR:OFF RTS:ON -> EN:0 IO0:1
2. Wait 100ms
3. DTR:ON RTS:OFF -> EN:1 IO0:0
4. Wait 100ms
5. DTR:ON RTS:ON -> EN:1 IO0:1 (release line)

As there are 2 lines need to be toggled simultaneously it's not possible to it currently.

I can make a PR either:
- Add a new menu entry
- In line toggle/pulse a new option for RTS&DTR

lundmar, 2024-03-29:
This is related to #219

We need to figure out how to add support for all these specific reset/boot sequences without having to maintain all sorts of hard coded boot sequences and pulse key options.

This is why I am now certain that adding script support for tio will be the best way to allow users to define whatever pulse sequence they need.

In your case the script option may look like this:
```
low(DTR); high(RTS); msleep(100); high(DTR); low(RTS); msleep(100); high(RTS);
```

Script can be run at boot and triggered via key in session.

lundmar, 2024-03-29:
> PS: CTS & DSR & CD in line operation could be removed as these signals are input only.

True, however, technically it is possible to set these via the serial port API so we kept those options. This is of course against the UART standards but for whatever reason the serial API allows it.

lundmar, 2024-04-01:
I've gone ahead and added Lua scripting support in 0becfa3274c80cb8d18fed0325c776eb4bfdb1ad.

I've also added the necessary Lua functions needed to manipulate the tty control lines in ways that should be able to support the reset or bootloader feature of any chipset. A very flexible solution.

You can define everything on the command line or use the configuration file like so:
```
[esp32]
device = /dev/ttyUSB0
color = 14
script = high(DTR); low(RTS); msleep(100); low(DTR); high(RTS); msleep(100); low(RTS)
#script-file = example/control-lines-test.lua
script-run = always
```

This configuration runs the script upon every connect ("always"). If needed you can reconfigure script-run to "once" to only run script at first connect. Use "never" to not run the script automatically in case you want to run the script manually using the ```ctrl-t r``` in session key code.

Also, instead of putting the script in the script variable you can instead use the script-file option and put the filename of whatever script file you want to run - this will be useful for more complex scripting in the future. If a script file is defined it will take priority and ignore anything defined in the script variable.

Feel free to test this new feature. I think it is an elegant solution to this issue.

My only concern regarding the implementation is the low/high vs off/on naming. I'm leaning towards low/high in case the user is connecting something where the UART on/off logic does not apply the same way.

HiFiPhile, 2024-04-01:
Thanks for your quick response !

I pulled the commit but I couldn't enter bootloader under Cygwin(msys) while under Linux is Ok, so I did some investigations:

With this script who toggle both line together: `high(RTS); high(DTR);msleep(100); low(DTR); low(RTS); msleep(100); high(RTS); high(DTR); msleep(100); low(RTS); low(DTR); msleep(100);`

Under Linux it works well, timing is nearly 100ms:
[image]

However under Windows the timing is ugly, pulse width become 156ms and there is a delta of 16-20ms between 2 lines.
[image]

Then I tested simple line pulse, I got a timing variation of 115-130ms....

---
I just done a hacky port (it'll hurts your eye...) to compile `tio` as a native win32 program to workaround Cygwin serial port limits, for line pulse I use one API call to set both lines and it works well. Although delay timing variation is still bad but at least 2 lines are changed at the same time.
https://github.com/HiFiPhile/tio/tree/win32

lundmar, 2024-04-01:
Thanks for testing. I'm glad the implementation works as intended on Linux. 👍🏻 

For Cygwin I guess I'm not too surprised that the timing is off considering that Cygwin is known to be slow.

Ha ha, yes - the hacky win32 port does hurt my eyes a little but I appreciate your effort to try get tio to run natively on win32.

Maybe there are some tricks to improve the delay precision on win32. Maybe try use another API instead of nanosleep() to see if that makes a difference. Maybe try increase the scheduling priority of tio etc.

HiFiPhile, 2024-04-02:
More experimentations, I changed the script handling by only saving high/low state when `low(xxx); high(x);` are called, then the actual line change is done at next sleep or script ending, to let one ioctl() call set all lines at the same time.
[line_mask.patch](https://github.com/tio/tio/files/14834191/line_mask.patch)

Even 100ms is still not precise (which is not critical in most cases), all lines are synchronized now:
[image]

lundmar, 2024-04-02:
Can you try revert your patch and only remove the tio_printf() calls? I suspect that the printouts may be the most taxing part of the code path.

HiFiPhile, 2024-04-02:
> I suspect that the printouts may be the most taxing part of the code path.

No luck, with printf() removed from `tty_line_set()` lines are still glitched about 15-18ms
[image]

lundmar, 2024-04-02:
Hmm, it must be the system calls that are most taxing in cygwin then. Lua is the fastest scripting language around and I simply don't believe that it nor the rest of the code path contributes milliseconds of delay.

Assuming that it is the system calls that are expensive then we can fix the issue by extending the tio Lua API with something like this which will result in only one system call:
```
script = set(RTS=low, DTR=high)
```

I'll see if I can make that syntax work in lua.

lundmar, 2024-04-06:
I couldn't quite make the Lua API as simple as I wanted above so I've introduced a tty line configuration script API to resolve the timing issue. See 3e5019110729be65e4c8d83e543c456c3786c02d

HiFiPhile, 2024-04-06:
Thanks for your update, now it works well !

closed 2024-04-06 by HiFiPhile

lundmar, 2024-04-29:
> Assuming that it is the system calls that are expensive then we can fix the issue by extending the tio Lua API with something like this which will result in only one system call:
> 
> ```
> script = set(RTS=low, DTR=high)
> ```
> 
> I'll see if I can make that syntax work in lua.

In preparation for tio v3.0 I've managed to rework the line manipulation API to be closer to my original idea above which makes it much simpler. See ba78a7b1cf5c80e4494b3de72c694d6b809f1c85.

lundmar, 2024-05-10:
> I just done a hacky port (it'll hurts your eye...) to compile `tio` as a native win32 program to workaround Cygwin serial port limits, for line pulse I use one API call to set both lines and it works well.

FYI - I recently switched tio to github workflow for GNU/Linux and MacOS CI builds and I noticed that github also has windows build machines available. Also, tio now only depends on glib and liblua and as far as I understand both of these are available as native windows binaries.

My point is that I think now we have everything available to make tio work on windows in case you ever feel like transforming your hacky port into a proper up-streamed win32 port ;)

HiFiPhile, 2024-05-10:
> My point is that I think now we have everything available to make tio work on windows in case you ever feel like transforming your hacky port into a proper up-streamed win32 port ;)

Of cause that would be awesome :) Glibc and liblua are the easy parts, what's difficult is Windows has a completely different IO model (compared to everything is a file) I've to use libserialport to hide implementation details. Also message queue and polling are quite different.

lundmar, 2024-05-10:
I suspected that. If libserialport minimizes the code changes required for a win32 port then I am all for it.

HiFiPhile, 2024-05-10:
It uses LGPL-v3 license I think won't be an issue, I've used it for some time and it's reliable at least on Windows and Linux.

But RS485 part is missing maybe because it's only available on Linux.

lundmar, 2024-05-10:
I don't expect all features to be available on all platforms. The RS485 feature is gated by a HAVE_RS485 flag which is set if meson detects the required RS485 headers are available on Linux so for a win32 port it will just disable.

HiFiPhile, 2024-05-10:
Yeah it's possible to use libserialport for normal things, as port fd is exposed current RS485 code is still compatible.

You can try to make a test build to see if it fits.

lundmar, 2024-05-10:
I don't want to use libserialport for the current implementation as it has been tested for years. I think that a win32 port can use libserialport. Also, if it makes sense a porting layer can be introduced but such decision is easier to make when it is known what changes are required for a win32 port.

HiFiPhile, 2024-05-10:
>  Also, if it makes sense a porting layer can be introduced but such decision is easier to make when it is known what changes are required for a win32 port.

A porting layer is welcomed, basically all posix functions need to be platform independent, go though file by file:
- termios.h
- unistd.h has very limited functions
- ENV different in config_file_resolve
- errno
- sys/ioctl.h
- sys/poll.h
- socket, may need a new thread
- Serial, RS485, character mapping
- Communication pipe
- Port enumerate, all devices of the same type have the same name (eg. `USB Serial Device` for ttyACM)

lundmar, 2024-05-10:
Right. I also just realize I have to clean up the headers. There has been a lot of copy pasting of headers going on.

HiFiPhile, 2024-05-30:
Hi, do you have any progress about the porting layer ?

lundmar, 2024-05-30:
@HiFiPhile I'm sorry. I'm a bit busy with other projects at the moment so I have to put a porting layer the TODO list for now.