tio/tio #234: Ambiguous line level

HiFiPhile, 2024-03-29:
Due to UART use active-low signaling, the HIGH/LOW level displayed are inverted, LOW is actually 3.3v and HIGH is 0v.

Either:
- Invert HIGH/LOW display
- Use ON/OFF instead

lundmar, 2024-03-29:
I'll have to look into this. We are asking serial port subsystem for the level and if inverted we will have to fix that. However, I think this may depend on the serial port driver. I can imagine that some invert this logic, meaning it may depend on the HW used whether or not the level is reported accurately.

HiFiPhile, 2024-03-29:
> However, I think this may depend on the serial port driver.

I believe drivers return logical value, 0 in idle and 1 activated. UART protocol is well defined as high level idle.

At least it's the case for FTDI, CP2102, and CH343 driver on both Windows and Linux.

lundmar, 2024-03-30:
Fixed in 6fee8514f425e4148577bc55878fbb28e41b7679

closed 2024-03-30 by lundmar