tio/tio #235: Add ability to send a BREAK over the socket

pcc, 2024-04-03:
I recently had a need to send a BREAK over the socket. Since I never need to send a NUL I did it using a local patch that replaces NUL with BREAK.
```
diff --git a/src/tty.c b/src/tty.c
index a1e4233..d1beb7a 100644
--- a/src/tty.c
+++ b/src/tty.c
@@ -992,7 +992,14 @@ void forward_to_tty(int fd, char output_char)
         {
             /* Send output to tty device */
             optional_local_echo(output_char);
-            status = tty_write(fd, &output_char, 1);
+            if (output_char == 0)
+            {
+                status = tcsendbreak(fd, 0);
+            }
+            else
+            {
+                status = tty_write(fd, &output_char, 1);
+            }
             if (status < 0)
             {
                 tio_warning_printf("Could not write to tty device");
```
With that, it becomes possible to e.g. reboot a Linux target using the Magic SysRq like so:
```
echo -ne '\0b' | nc -UN /tmp/my-sock
```
But ideally there would be a way to do this without losing the ability to send a NUL (or some other character). To begin with, perhaps we can introduce an `ONULBRK` option which triggers this behavior. Then if someone needs to be able to send both NUL and BREAK over the same socket, we could introduce another option that e.g. makes it so that NUL becomes an escape character so you send another NUL over the socket to send a NUL over the serial line or a 'b' to send a BREAK.

pcc, 2024-04-05:
One advantage of the single socket is we get guarantees "for free" about orderings between characters. With the control socket I am unsure whether tio can provide a guarantee to the client that BREAKs are sent in the correct order wrt other characters. For example, when sending the Magic SysRq reboot command, the BREAK needs to be transmitted before the 'b'. If the client program has both control and data sockets open, sends a BREAK on the control socket and a 'b' on the data socket, tio could end up reading the data socket first and transmitting the 'b' first.

It may be possible to enforce an ordering by requiring the client to open and close sockets in a specific order. For example, to send a SysRq reboot, maybe the client would need to do something like this:
1. open data socket
2. [do any other communication that needs to happen before the reboot]
3. open control socket
4. send BREAK to control socket
5. close control socket
6. send 'b' to data socket

But this still requires tio to see 5 before 6, which I'm not sure is guaranteed. Or we could require closing data socket between 2 and 3 and re-opening data socket between 5 and 6, which seems more likely to be guaranteed to work but I haven't thought it through completely yet and it would be more tricky to manage in a program like expect, as well as creating the possibility of missing received characters.

lundmar, 2024-04-05:
We best support ordering so I've gone ahead and implemented the simple solution. Implemented in d9be6ee7fe4b59770f4f459e7e0a924b18b17a78.

closed 2024-04-05 by lundmar