tio/tio #238: tio isn't a simple serial device I/O tool any more.

yangh, 2024-04-11:
More and more feature added to tio and it's not a simple tool any more.

lundmar, 2024-04-11:
Right you are. Tio has evolved to become more of a tool for professional developers and hackers. Though, still simple to use for most common use cases.

yangh, 2024-04-11:
Hi Martin

Thanks for your feedback, let's looking forward the future of tio, we close this ticket (not an issue actually) any time.

BR.

lundmar, 2024-04-11:
No worries. I'm still trying to keep things as simple as possible.

In the start I created tio to just be an ultra simple alternative to screen for connecting to serial ports of my embedded boards. However, after some years I had to give up on the ultra simple idea because people kept suggesting and contributing new features that made sense, especially for embedded. This lead me to the realization that there is a need for a serial tool with more focus on the needs of developers and hackers. Tio is not your classic serial modem tool.

Hence my responsibility now is to maintain tio and accept features that make sense for embedded  while keeping the user interface and implementation as simple and clean cut as possible.

closed 2024-04-11 by lundmar