tio/tio #239: Disable DEC Special Graphics at exit

pcc, 2024-04-16:
The DEC Special Graphics charset is activated by sending a Shift In character (`\016`) and causes most terminal emulators to replace lower case characters with line drawing characters. With serial adapters, this character is most commonly seen as part of line noise. If tio receives the Shift In character, the terminal will remain in Special Graphics mode when the program exits. To mitigate this, we should consider printing the Shift Out character `\017` at exit and before each log message.

lundmar, 2024-04-16:
If I understand correctly this is only a problem for vt100 users. I don't think many modern terminal emulators defaults to that anymore.

We could implement a mechanism that detects it on the tty device input stream and only then do what you suggest. That will take care of the line noise scenario.

pcc, 2024-04-16:
Ah, I thought it was a universal feature (I remember being annoyed by this behavior for decades and only now decided to look into what causes it) but I spot checked some terminals (gnome-terminal, xterm, WezTerm) and it turns out that of those, only WezTerm implements it by default, which is the terminal emulator that I use. Sending a `\017` only after seeing a `\016` would work for me.

lundmar, 2024-04-27:
Implemented in f5f62ee02dbfc43c18262517f3611ccb91147288. Will be part of upcoming tio v3.0.

lundmar, 2024-04-27:
Feel free to test.

closed 2024-05-02 by lundmar