tio/tio #241: Broken on Windows?

FerriteGiant, 2024-04-17:
I installed via `$ pacman -S tio` which worked fine. But it doesn't connect or see any COM ports.

I tried running it in the MSYS console and in powershell after adding the bin directory to my windows path.

In both cases, `--list-devices` returns nothing and `tio COMXXX` says "Warning: Could not open tty device (No such file or directory)"

Note that I am able to connect to the COM port (which is just a serial-to-USB adapter) with PuTTY with no issue.

This could totally be user error, but in that case probably should add more details to the readme.

lundmar, 2024-04-18:
It should be working. Lately I see that @HiFiPhile is using tio successfully in a cygwin/msys setup connecting to "COM10": https://github.com/tio/tio/issues/233#issuecomment-2031674384

Perhaps try look for /dev/ttyS* devices and try to connect to those.

lundmar, 2024-04-18:
Oh, I just realize that the "COM10" string may just be the name of his sub configuration and so it probably sets device to /dev/ttyS10 .

FerriteGiant, 2024-04-18:
[image]
 
 This is what I see both with and without the device connected

FerriteGiant, 2024-04-18:
If I connect to /dev/tty it says connected, but doesn't actually show anything

lundmar, 2024-04-18:
AFAIK the device should be named /dev/ttySN in cygwin/msys. So your problem is that you don't have any such device. I think something is broken in your msys/cygwin setup.

I assume putty works because you are not running it under the cygwin/msys environment.

lundmar, 2024-04-18:
> If I connect to /dev/tty it says connected, but doesn't actually show anything

This is not a serial port tty device.

FerriteGiant, 2024-04-18:
> > If I connect to /dev/tty it says connected, but doesn't actually show anything
> 
> This is not a serial port tty device.

Yeah, I know. Just tried for completeness

FerriteGiant, 2024-04-18:
> AFAIK serial devices should be named /dev/ttySN in cygwin/msys. So your problem is that you don't have any such device. I think something is broken in your msys/cygwin setup.
> 
> I assume putty works because you are not running it under the cygwin/msys environment.

Do I need to have Cygwin installed in addition to MSYS?

lundmar, 2024-04-18:
No, MSYS is an alternative to Cygwin.

I just remember that the reason that you do not see the serial usb device maybe be because you are not running msys terminal with administrative rights.

FerriteGiant, 2024-04-18:
Hmm, tried all of those to no avail. I'll play around with it some more later and let you know if I figure it out.

FerriteGiant, 2024-04-18:
Well on a different Windows 10 computer the serial devices show up just fine. However I am hitting another issue, which is that anytime the device disconnects there is a permission error. (I'm running as admin and I tried the 32bit and 64bit consoles.)

[image]

lundmar, 2024-04-18:
You can ignore the exit error. It is a know issue that has been fixed in tio v2.8. See #205 .

You can try request the msys guys to update tio.

closed 2024-04-18 by lundmar