tio/tio #243: XMODEM added modes and menu in tio 2.8

eshattow, 2024-04-20:
Good to see added XMODEM variants in tio 2.8 ref: #236

Concerned about the insertion of a menu into UX workflow. Put simply it is annoying to add an extra step to the existing workflow for a very commonly used option.

If accepting that a menu must be inserted, then asking for mode as a menu selection before the filename is weird UX as this might fail regardless of the forced menu selection if the named file is not accessible.

Was it not acceptable to have `ctrl+t x` for default mode and `ctrl+t X` to select the desired XMODEM variant, or as a more general menu driven list of all transfer modes and variants (including YMODEM, XMODEM)?   Retain existing 'x' and 'y' shortcuts as these are most common.

Otherwise is there some desire to streamline and remove `ctrl+t y` YMODEM into a more generic menu-driven upload shortcut and eliminate the convenient XMODEM  and YMODEM shortcuts?

Appreciating the new functionality but not sure what to suggest, it is named in documentation for setting up single board computers in a few external places how to use tio and `ctrl+t x` which now suddenly doesn't work as before.

HiFiPhile, 2024-04-20:
> If accepting that a menu must be inserted, then asking for mode as a menu selection before the filename is weird UX as this might fail regardless of the forced menu selection if the named file is not accessible.

I don't think so, all the tools I used ask mode first then file name.

> or as a more general menu driven list of all transfer modes and variants (including YMODEM, XMODEM)? Retain existing 'x' and 'y' shortcuts as these are most common.

I like the idea to group all modems, somebody may implement Zmodem for example.

> Retain existing 'x' and 'y' shortcuts as these are most common.

It's hard to define the popularity for Xmodem, in my 10 years electronic engineer career I've hardly used Xmodem-1k, either products use Xmodem-CRC or Ymodem if 1k block size is used.

lundmar, 2024-04-20:
> > or as a more general menu driven list of all transfer modes and variants (including YMODEM, XMODEM)? Retain existing 'x' and 'y' shortcuts as these are most common.
> 
> I like the idea to group all modems, somebody may implement Zmodem for example.

It is a balancing act. I like the new sub menu because it provides category structure and avoids cluttering top level key commands. However I also don't want sub menu to go too deep so I'm reserving ctrl-t x,y,z for z/x/y modem key commands. I think this makes sense.

lundmar, 2024-04-20:
> Appreciating the new functionality but not sure what to suggest, it is named in documentation for setting up single board computers in a few external places how to use tio and `ctrl+t x` which now suddenly doesn't work as before.

I'm sorry it breaks your documentation but tio is constantly improving and so is the UX, be it key commands or command line options etc. For example, recently we removed options because the new scripting feature made them obsolete.

I appreciate your strong opinions on the key command UX but I think the new sub menu is an improvement for the reasons stated.

That being said I just want to note that I'm glad that people have put in the effort to add native modem transfer features instead of calling external programs like sx,sz, etc. Big thanks to @pnrhub and @HiFiPhile .

closed 2024-04-27 by lundmar