tio/tio #247: "error: storage size of 'stx' isn't known" and "error: 'STATX_ALL' undeclared"

robert-scheck, 2024-05-05:
Building tio v3.1 on EPEL 7 for RHEL/CentOS 7 fails like this:

```
[7/18] cc -Isrc/tio.p -Isrc -I../src -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -pipe -D_FILE_OFFSET_BITS=64 -std=gnu99 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -pthread -Wno-unused-result -DHAVE_TERMIOS2 -DHAVE_RS485 -MD -MQ src/tio.p/fs.c.o -MF src/tio.p/fs.c.o.d -o src/tio.p/fs.c.o -c ../src/fs.c
FAILED: src/tio.p/fs.c.o 
cc -Isrc/tio.p -Isrc -I../src -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -pipe -D_FILE_OFFSET_BITS=64 -std=gnu99 -O2 -g -Wall -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m64 -mtune=generic -pthread -Wno-unused-result -DHAVE_TERMIOS2 -DHAVE_RS485 -MD -MQ src/tio.p/fs.c.o -MF src/tio.p/fs.c.o.d -o src/tio.p/fs.c.o -c ../src/fs.c
../src/fs.c: In function 'fs_get_creation_time':
../src/fs.c:186:18: error: storage size of 'stx' isn't known
     struct statx stx;
                  ^
../src/fs.c:193:5: warning: implicit declaration of function 'statx' [-Wimplicit-function-declaration]
     if (statx(fd, "", AT_EMPTY_PATH, STATX_ALL, &stx) != 0)
     ^
../src/fs.c:193:38: error: 'STATX_ALL' undeclared (first use in this function)
     if (statx(fd, "", AT_EMPTY_PATH, STATX_ALL, &stx) != 0)
                                      ^
../src/fs.c:193:38: note: each undeclared identifier is reported only once for each function it appears in
../src/fs.c:186:18: warning: unused variable 'stx' [-Wunused-variable]
     struct statx stx;
                  ^
../src/fs.c:203:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
```

And yes, RHEL/CentOS 7 is quite old (e.g. glibc 2.17), also it reaches EOL at 2024-06-30, which is in nearly two months. Just let me know if your answer is "won't fix", because too old Linux distribution.

sodface, 2024-05-05:
I had this same issue building for Alpine Linux locally on 3.19.1, which is the most current stable version. The problem went away when I switched the build to edge (future Alpine 3.20), which is also what the Gitlab CI builds on so I didn't look into it any further.

lundmar, 2024-05-05:
@robert-scheck Should be fixed in b490233988ebe9dbc015690cd4cde9acd2e5e1e1 so that it falls back to using less precise creation time on older GNU/Linux systems without statx().

Just use it as a build patch. I don't expect to make another tio release anytime soon. I think I have added all the features I want for now! 😃

robert-scheck, 2024-05-05:
> @robert-scheck Should be fixed in [b490233](https://github.com/tio/tio/commit/b490233988ebe9dbc015690cd4cde9acd2e5e1e1) so that it falls back to using less precise creation time on older GNU/Linux systems without statx().
> 
> Just use it as a build patch. I don't expect to make another tio release anytime soon. I think I have added all the features I want for now! 😃

Yes, that works for EPEL 7 - thank you.

closed 2024-05-05 by robert-scheck