tio/tio #250: TIO snap install on Ubuntu Unity 24.04 is not working due to GLIBC

KhazAkar, 2024-05-20:
Hi,
after installing tio using snap on my Ubuntu Unity 24.04 install, it doesn't work.
Current error is:
```
/snap/tio/1097/usr/bin/tio: /snap/core20/current/lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /lib/x86_64-linux-gnu/libgtk3-nocsd.so.0)
```

lundmar, 2024-05-20:
The snap recipe is right here https://github.com/tio/tio.snapcraft

The tio snap works fine on Ubuntu 24.04 so the Unity flavor must be doing something different.

KhazAkar, 2024-05-22:
Did my investigation. In order to fix it, you need to remove 2 packages:
`gtk3-nocsd libgtk3-nocsd0`
Then tio starts.
Thanks for the link to tio snapcraft recipe, will definitely play with it, like upgrading core :)

closed 2024-05-22 by KhazAkar

lundmar, 2024-05-22:
Feel free to make a PR that does all of the above.