tio/tio #255: Tio script "send" command not working from command line

CF209, 2024-06-04:
I'm trying to run a script when starting tio. It appears to work when I put the script in my config file, but it doesn't work when I run it from the command line.

Here's the profile in my config file which works when I run `tio reboot`:
```
[reboot]
device = /dev/cu.usbmodem2102
script = send("BOUFFALOLAB5555DTR0"); msleep(100); send("BOUFFALOLAB5555RTS0"); msleep(100); send("BOUFFALOLAB5555RTS1");
```

Here's the command I'm running that doesn't work though:
```
tio --script "send("BOUFFALOLAB5555DTR0"); msleep(100); send("BOUFFALOLAB5555RTS0"); msleep(100); send("BOUFFALOLAB5555RTS1");" /dev/cu.usbmodem2102
```

lundmar, 2024-06-04:
You have to be careful when using single or double quotes on the command line.

Try this instead:
```
tio --script 'send("BOUFFALOLAB5555DTR0"); msleep(100); send("BOUFFALOLAB5555RTS0"); msleep(100); send("BOUFFALOLAB5555RTS1");' /dev/cu.usbmodem2102
```

CF209, 2024-06-04:
Ah, I didn't even think about that since the sleep commands were still working. Thanks!

closed 2024-06-04 by CF209