tio/tio #260: Request:  mode for omitting CR to output

LouisLambda, 2024-06-25:
I'm working with a particularly brain dead board, which has a very snooty command parser.

I need a way to configure tio so that:
1. what I type in is not sent until I hit enter (i.e. input line mode)
2. The resulting '\r' triggers submission but is NOT actually sent over the wire.

I can't find a way to configure this. It's not `-m IGNCR`, or anything else I tried.
tio always sends the '\r' shortly after sending the input buffer (from `forward_to_tty`),  and this causes the other side of the link to treat the command I just sent as invalid.

Adding 
```
    if ((output_char == '\r'))
    {
        return;
    }
```
to the start of `forward_to_tty` does what I need. If there was a `-m` option I could specify to get this behaviour, I'd be happy.

Thanks

lundmar, 2024-06-25:
Hmm, that is a very special parser. This means you are just feeding the board lines without any CR or NL to terminate? The board knows when you have entered something of value?

LouisLambda, 2024-06-25:
That's exactly right.  The interface was designed for Microcontrollers, but for debug I needed tio to accommodate it.

lundmar, 2024-06-25:
Okay. So what you need is basically a OIGNCR

lundmar, 2024-06-25:
IGNCR means that you will se no CR on input from tty device. OIGNCR will mean that any CR will not be sent to tty device.

We can make it so that in line mode with local echo enabled it will still go to the next line but it will not send the CR character to the device.

LouisLambda, 2024-06-25:
That sounds perfect.

lundmar, 2024-06-27:
Add support for OIGNCR mapping flag in 4723cc3f4e3e8c3ca1ff1580cf8767a837afabbb. Feel free to test.

LouisLambda, 2024-06-28:
I tested and it's working perfectly. Thanks!

closed 2024-06-28 by LouisLambda