tio/tio #261: Another kind of timestamp

saqwed, 2024-06-28:
Hi,

I know tio already support timestamp feature like below style

```
...
[18:13:07.488] localhost login:
[18:13:08.557] localhost login:
[18:13:09.635] localhost login:
```
and I can calculate the first two lines pass about 1.069s and the next gap is 1.078s.
just like to know is it possible to have another timestamp style, for example

```
...
[??:??:??.???] localhost login:
[00:00:01.069] localhost login:
[00:00:01.078] localhost login:
```

so I can easily to know the time gap for each line, thank you.

lundmar, 2024-06-28:
Tio already supports delta timestamps. Look for 24hour-delta in the tio man page.

closed 2024-06-28 by lundmar

saqwed, 2024-06-28:
oops, sorry, i didn't aware the man link at [man](https://[raw.githubusercontent.com/tio/tio/master/man/tio.1.txt](https://raw.githubusercontent.com/tio/tio/master/man/tio.1.txt))

I just type `man tio` directly and no help so question here.
thanks for your quick reply.

```
$ man tio
No manual entry for tio
```