tio/tio #262: Tio list not helpful on MacOs

KindDragon, 2024-07-01:
tio --list return:
```
tio --list                                                                                                                                            ✔  2m 18s  multiverse   18:48:29 
Device            TID     Uptime [s] Driver           Description
----------------- ---- ------------- ---------------- --------------------------
/dev/ttyp0             1719848914.615                  
/dev/ptyp0             1719848914.615                  
/dev/ttyp1             1719848914.615                  
/dev/ptyp1             1719848914.615                  
/dev/ttyp2             1719848914.615                  
/dev/ptyp2             1719848914.615                  
/dev/ttyp3             1719848914.615                  
/dev/ptyp3             1719848914.615                  
/dev/ttyp4             1719848914.615                  
/dev/ptyp4             1719848914.615                  
/dev/ttyp5             1719848914.615                  
/dev/ptyp5             1719848914.615                  
/dev/ttyp6             1719848914.615                  
/dev/ptyp6             1719848914.615                  
/dev/ttyp7             1719848914.615                  
/dev/ptyp7             1719848914.615                  
/dev/ttyp8             1719848914.615                  
/dev/ptyp8             1719848914.615                  
/dev/ttyp9             1719848914.615                  
/dev/ptyp9             1719848914.615                  
/dev/ttypa             1719848914.615                  
/dev/ptypa             1719848914.615                  
/dev/ttypb             1719848914.615                  
/dev/ptypb             1719848914.615                  
/dev/ttypc             1719848914.615                  
/dev/ptypc             1719848914.615                  
/dev/ttypd             1719848914.615                  
/dev/ptypd             1719848914.615                  
/dev/ttype             1719848914.615                  
/dev/ptype             1719848914.615                  
/dev/ttypf             1719848914.615                  
/dev/ptypf             1719848914.615                  
/dev/ttyq0             1719848914.615                  
/dev/ptyq0             1719848914.615                  
/dev/ttyq1             1719848914.615                  
/dev/ptyq1             1719848914.615                  
/dev/ttyq2             1719848914.615                  
/dev/ptyq2             1719848914.615                  
/dev/ttyq3             1719848914.615                  
/dev/ptyq3             1719848914.615                  
/dev/ttyq4             1719848914.615                  
/dev/ptyq4             1719848914.615                  
/dev/ttyq5             1719848914.615                  
/dev/ptyq5             1719848914.615                  
/dev/ttyq6             1719848914.615                  
/dev/ptyq6             1719848914.615                  
/dev/ttyq7             1719848914.615                  
/dev/ptyq7             1719848914.615                  
/dev/ttyq8             1719848914.615                  
/dev/ptyq8             1719848914.615                  
/dev/ttyq9             1719848914.615                  
/dev/ptyq9             1719848914.615                  
/dev/ttyqa             1719848914.615                  
/dev/ptyqa             1719848914.615                  
/dev/ttyqb             1719848914.615                  
/dev/ptyqb             1719848914.615                  
/dev/ttyqc             1719848914.615                  
/dev/ptyqc             1719848914.615                  
/dev/ttyqd             1719848914.615                  
/dev/ptyqd             1719848914.615                  
/dev/ttyqe             1719848914.615                  
/dev/ptyqe             1719848914.615                  
/dev/ttyqf             1719848914.615                  
/dev/ptyqf             1719848914.615                  
/dev/ttyr0             1719848914.615                  
[... 174 lines cut ...]
/dev/ptyw7             1719848914.615                  
/dev/ttyw8             1719848914.615                  
/dev/ptyw8             1719848914.615                  
/dev/ttyw9             1719848914.615                  
/dev/ptyw9             1719848914.615                  
/dev/ttywa             1719848914.615                  
/dev/ptywa             1719848914.615                  
/dev/ttywb             1719848914.615                  
/dev/ptywb             1719848914.615                  
/dev/ttywc             1719848914.615                  
/dev/ptywc             1719848914.615                  
/dev/ttywd             1719848914.615                  
/dev/ptywd             1719848914.615                  
/dev/ttywe             1719848914.615                  
/dev/ptywe             1719848914.615                  
/dev/ttywf             1719848914.615                  
/dev/ptywf             1719848914.615                  
/dev/ptmx              1719848914.615                  
/dev/cu.BoseQC35S2      1719848914.615                  
/dev/tty.BoseQC35S2      1719848914.615                  
/dev/cu.Bluetooth-Incoming-Port      1719848914.615                  
/dev/tty.Bluetooth-Incoming-Port      1719848914.615                  
/dev/cu.usbserial-0001      1719848914.615                  
/dev/tty.usbserial-0001      1719848914.615
```

lundmar, 2024-07-01:
5f70b75e9053c2651525f0b8dd43c6bbc84ff1a3 should shorten the listing on macOS - feel free to test.

lundmar, 2024-07-08:
I'm not sure what is going on with the calculated uptime. AFAIK the functions used in the calculation ( get_current_time() and fs_get_creation_time() ) should work just fine on MacOS. Unfortunately I can't debug it since I do not have a mac.

KindDragon, 2024-07-09:
> I've investigated the code and AFAIK the functions involved ( get_current_time() and fs_get_creation_time() ) should work on macOS but clearly do not.

I think you should modification time on MacOs instead
[image]

KindDragon, 2024-07-09:
I also implemented a prototype (using ChatGPT :) because I don't know anything about IOKit) to extract additional information from MacOs about serial devices https://github.com/KindDragon/tio/commit/b2f157d28bdef801ad8162d619d0a010f7b1ad2e

```
Device                         Port Location    TID  productName vendorName
----------------------------- --------------- ---------------- --------------------------
/dev/tty.BoseQC35S2                 Unknown             'Unknown         ' 'Unknown         '
/dev/tty.Bluetooth-Incoming-Port    Unknown             'Unknown         ' 'Unknown         '
/dev/tty.usbserial-21310            02131000        Ohyf 'CP2102N USB to UART Bridge Controller' 'Silicon Labs    '
/dev/tty.usbserial-0001             02120000        0pqa 'CP2102 USB to UART Bridge Controller' 'Silicon Labs    '
```

lundmar, 2024-07-09:
> Next up: Do you care if it works on Catalina (MacOS 10.15.7)? I thought a trivial fix would be enough, but that needs some team debugging, too. Right now, I'm off to grab the M1 and confirm top-of-tree working there.

Sure. I mean, unless it is an extremely old system we can take care of it. Feel free to open an new issue for that.

lundmar, 2024-07-09:
> I also implemented a prototype (using ChatGPT 😅 because I don't know anything about IOKit) to extract additional information from MacOs about serial devices [KindDragon@b2f157d](https://github.com/KindDragon/tio/commit/b2f157d28bdef801ad8162d619d0a010f7b1ad2e)
> 
> ```
> Device                         Port Location    TID  productName vendorName
> ----------------------------- --------------- ---------------- --------------------------
> /dev/tty.BoseQC35S2                 Unknown             'Unknown         ' 'Unknown         '
> /dev/tty.Bluetooth-Incoming-Port    Unknown             'Unknown         ' 'Unknown         '
> /dev/tty.usbserial-21310            02131000        Ohyf 'CP2102N USB to UART Bridge Controller' 'Silicon Labs    '
> /dev/tty.usbserial-0001             02120000        0pqa 'CP2102 USB to UART Bridge Controller' 'Silicon Labs    '
> ```

Ha ha, nice. I was hoping someone would maybe try fill in the description field implementation so the list output becomes a bit more useful if not as useful as the one on Linux.

I see you are changing the output fields compared to the original. I suggest you keep the columns the same but combine the product name and vendor name and use that as the "Description".

lundmar, 2024-07-09:
> I have a variety of WCH, SiLabs, Prolific, and other controllers if we want to try to fill them in.

The more testing the better.

> IIRC, I've written code in the past that groped IOKit for this info and found it wildly variable between device drivers. I tried to use it to populate menus and remember I ended up not using productname exactly for reasons like that above - "Unknown" isn't so helpfu.

Let's inspect the output from some of your devices and see how those names fair.

KindDragon, 2024-07-11:
I couldn't see /dev/tty.E500, /dev/tty.Bluetooth-Incoming-Port devices in your ioreg output. Could you provide `ioreg -r -c IOSerialBSDClient -x -l` results?

lundmar, 2024-07-17:
I want to make a new tio release soon so we can get the recent macOS fixes deployed.

Can one of you guys please confirm that the tio listing looks good, in particular that the uptime is now working correctly on macOS (non-Catalina).

Thanks.

lundmar, 2024-07-18:
@robertlipe Sorry, I didn't make it clear. Can you test mainline without KindDragon's changes?

@KindDragon Once you make a PR we can get your listing feature included. However, I think we need to make the output look like the one on linux, meaning having the same named table columns.

lundmar, 2024-07-18:
> Note that mixing list and -t is a mess.

That should not happen. Works fine with the Linux build.

lundmar, 2024-07-18:
That is what I expected it would look like except the double ```----``` stuff? Is that a copy paste error or does it really look like that?

lundmar, 2024-07-18:
I'm okay with that for now for a relase. I just wanted to make sure that it is functional and correctly aligned. When ready, with KindDragon's feature we can fill in the description and maybe TID and maybe do away with the driver field.

lundmar, 2024-07-19:
@KindDragon I went ahead and made a new release to get the latest macOS fixes deployed. Hopefully for next release we can include your implementation. A PR will be appreciated :)

KindDragon, 2024-07-20:
Sorry, I was busy. 
@robertlipe could you provide parent devices for /dev/tty.E500, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.wlan-debug?
I think this command could do this `ioreg -r -c IOSerialBSDClient -x -l -t`
I want understand of it possible to get some additional information for them

KindDragon, 2024-07-21:
Thanks, I will check!

closed 2025-05-15 by KindDragon