tio/tio #266: Cannot send newline when output mode is hex

detly, 2024-07-10:
I'm connected to a device running OpenWRT. In the default input/output mode:

```text
# ls -1
user_ssh_keys
#
```

(Plus some ANSI color codes in there that didn't copy/paste across.)

If I do <kbd>ctrl</kbd>+<kbd>t</kbd>, <kbd>o</kbd> and then type `ls -1` then <kbd>enter</kbd>, I get nothing back. No characters are displayed whatsoever, not even the `ls -1`. The cursor does not move. (No, the input mode is not accidentally set to "line mode", I checked :wink:)

Interestingly, if I start from a `#` prompt again, type `ls -1` but _don't_ hit enter, change _both_ input and output modes to hex, then type `0A`... I get

```text
0d 0a 1b 5b 31 3b 33 34 6d 75 73 65 72 5f 73 73 68 5f 6b 65 79 73 1b 5b 6d 0d 0a 23 20
```

...which is

```text

[1;34muser_ssh_keys[m
# 
```

Configuration:

```text
Configuration:
 Device: /dev/ttyUSB0
 Baudrate: 115200
 Databits: 8
 Flow: none
 Stopbits: 1
 Parity: none
 Local echo: false
 Timestamp: none
 Timestamp timeout: 200
 Output delay: 0
 Output line delay: 0
 Automatic connect strategy: direct
 Automatic reconnect: false
 Pulse duration: DTR=100 RTS=100 CTS=100 DSR=100 DCD=100 RI=100
 Input mode: normal
 Output mode: normal
 Alert: none
 Mappings: none
```

Versions:
- tio: v3.6 (installed from Snap)
- OS: Ubuntu 24.04 (noble)

lundmar, 2024-07-10:
Thanks for reporting the issue. Fixed in c3654486c79d323d6a2485283897df77cbbc4cf4

Feel free to test. Thanks.

lundmar, 2024-07-10:
FYI - you can test by installing the snap edge version which should include the fix:

```
$ snap refresh tio --edge
```

lundmar, 2024-07-11:
I consider this issue fixed. Feel free to reopen if it does not work for you.

closed 2024-07-11 by lundmar

detly, 2024-09-02:
Sorry for the delay, I was away. But I can confirm your fix. Thanks.