tio/tio #268: Incorrect timestamp after X-Modem or Y-Modem transfer

xypron, 2024-07-13:
I am using tio  v2.7 as packaged in Ubuntu 24.04.

When transferring a file I saw this output:

```
[13:17:26.226] Send file with XMODEM
[13:17:33.168] Sending file 'spl/u-boot-spl.bin.normal.out'  
[13:17:33.168] Press any key to abort transfer
................................................................................................................................................|
[13:17:33.168] Done
```
In the `Done` line I would expect a timestamp that matches the termination of the transfer and not the start.

xypron, 2024-07-13:
Changes like the following are needed:

```c
diff --git a/src/tty.c b/src/tty.c
index c7f2e73..72cb515 100644
--- a/src/tty.c
+++ b/src/tty.c
@@ -1094,9 +1094,12 @@ void handle_command_sequence(char input_char, char *output_char, bool *forward)
                 tio_printf("Send file with YMODEM");
                 tio_printf_raw("Enter file name: ");
                 if (tio_readln()) {
+                    int ret;
+
                     tio_printf("Sending file '%s'  ", line);
                     tio_printf("Press any key to abort transfer");
-                    tio_printf("%s", xymodem_send(device_fd, line, YMODEM) < 0 ? "Aborted" : "Done");
+                    ret = xymodem_send(device_fd, line, YMODEM);
+                    tio_printf("%s", ret < 0 ? "Aborted" : "Done");
                 }
                 break;
```

Now I see that `Done` occurred later than `Sending file`:

```
[13:48:29.692] Send file with YMODEM
[13:48:32.788] Sending file 'u-boot.itb'  
[13:48:32.788] Press any key to abort transfer
.......................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................|
[13:50:15.832] Done
Loaded 1037679 bytes
```

lundmar, 2024-07-13:
Thanks for reporting the bug. Fixed in 37a4f33e87470524125e6ae4a29fd35b02a384d1

PR #269 (2024-07-13): Print correct 'Done' timestamp for X- and Y-modem transfers
Closes: #268

Call tio_printf() after completing xymodem_send().
commit 68a64ac5 (2024-07-13): Print correct 'Done' timestamp for X- and Y-modem transfers

Closes: #268

Call tio_printf() after completing xymodem_send().
closed 2024-07-13 by lundmar