tio/tio #271: Support for multiple config files

jwillikers, 2024-07-14:
tio has been incredibly useful for me to keep track of various serial device connections through the config file. Right now, I only use the config file for personal projects, but I'd like to have a completely separate config file for my work-related projects which I can share with my co-workers. This would require two separate config files, one for my personal projects and one for work. For my Git configuration, I accomplish this with an `include` for my work-related configuration file. Would it be possible to support something that would allow using multiple config files?

lundmar, 2024-07-14:
Yes, the configuration is very useful when working with various serial devices.

Your suggestion is good.

Unfortunately our configuration file parser (GKeyFile) does not support includes. However, I can implement it so that it is possible do something similar to gits config include feature.

For example:

```
[include work/config]
```

Which will include the file config located in the work directory relative to the location of the configuration file being parsed.

jwillikers, 2024-07-14:
That would work! Another approach would be a `conf.d` subdirectory like `~/.config/tio/conf.d/` where all files with a specific extension, usually `.conf`, are parsed in order.

lundmar, 2024-07-14:
The problem with the ```conf.d``` concept is that it loads all the files in the directory. This is good for service init etc. However if you have many configuration files but only want to include one specific file then you have to rename or move files around.

I think the include directive offers better flexibility. Then you can manage which specific files to include from the top level configuration file. It even makes it possible to have a hierarchy of active and passive configuration files - files including files etc.

lundmar, 2024-07-15:
FYI - I've added the feature to the TODO list.

jwillikers, 2024-07-15:
> FYI - I've added the feature to the TODO list.

Awesome, thank you!

lundmar, 2024-07-18:
Added in 4c39ece879dcd62036b136da4e3d64f6b8dcb759

Feel free to test.

closed 2024-07-19 by lundmar