tio/tio #272: Issues when connected to rPi headers directly

jordan-rash, 2024-08-02:
When connecting to my device to the rPi headers directly, I am having issues figuring out how to map returns correctly.

When directly connected:
```
$ tio -e --input-mode line -m ONLCRNL /dev/ttyAMA0
[13:57:39.509] tio v3.6
[13:57:39.509] Press ctrl-t q to quit
[13:57:39.511] Connected to /dev/ttyAMA0
AT

+ERR=4
+ERR=4
```

Also directly connected: 
```python3
import serial
import time

phone = serial.Serial("/dev/ttyAMA0", baudrate=115200, timeout=1.0)

cmd='AT\r\n'
phone.write(cmd.encode())
result=phone.read(100)
print(result)
```
$ python derp.py
b'+OK\r\n'               <- get the OK

When I put a usb serial converter on my device and use `tio`, and the same command above, I get the OK.  

With USB converter in place
```
$ tio -e --input-mode line -m ONLCRNL /dev/ttyACM0
[14:06:01.855] tio v3.6
[14:06:01.856] Press ctrl-t q to quit
[14:06:01.859] Connected to /dev/ttyACM0
AT

+OK

[14:06:07.736] Disconnected
```

Ultimately, i am just trying to use tio with the device directly connected to the rPi

jordan-rash, 2024-08-02:
I was able to get it to OK using the script functionality when directly connected
```
$ tio -e --script 'send("AT\r\n")' /dev/ttyAMA0
[14:19:38.969] tio v3.6
[14:19:38.970] Press ctrl-t q to quit
[14:19:38.971] Connected to /dev/ttyAMA0
[14:19:38.972] Running script
+OK

[14:19:40.276] Disconnected
```

lundmar, 2024-08-02:
Hmm, so the exact same tio configuration works with ttyACM0 but not ttyAMA0...

And you run these commands on the same system?

jordan-rash, 2024-08-02:
Same system...and same command, yes.  The only difference is how the device is connected (direct vs USB)....ill run the script and the direct back to back and post the differences

jordan-rash, 2024-08-02:
So here is the script working and direct not working in the same session
```
$ tio -e --input-mode line -m ONLCRNL --script 'send("AT\r\n")' /dev/ttyAMA0
[14:39:26.452] tio v3.6
[14:39:26.453] Press ctrl-t q to quit
[14:39:26.454] Connected to /dev/ttyAMA0
[14:39:26.455] Running script
+OK
AT

+ERR=4
+ERR=4
```

lundmar, 2024-08-02:
I take it you are running tio on the rpi itself?

jordan-rash, 2024-08-02:
yes

lundmar, 2024-08-02:
Can you try without line mode etc.?
`$ tio -m ONLCRNL /dev/ttyAMA0`

jordan-rash, 2024-08-02:
Every keystroke is returned as ERR=1 (invalid command) 😞 
```
$ tio -m ONLCRNL /dev/ttyAMA0
[14:51:19.451] tio v3.6
[14:51:19.451] Press ctrl-t q to quit
[14:51:19.453] Connected to /dev/ttyAMA0
+ERR=1
+ERR=1
+ERR=1
+ERR=1
```

lundmar, 2024-08-02:
You are firing the exact same commands and the only difference is AMA0 vs ACM0. So apparently something in the UART HW makes a difference. My initial thought was that perhaps tio does not sync/flush its buffer when pressing enter and that somehow affects AMA0 but not ACM0. However, I just checked the code and tio does indeed force a sync/flush so it should all have been shifted out on the UART.

This is puzzling...

jordan-rash, 2024-08-02:
Well we can remove the ACM0 from the equation even as the OK comes back _if I use your built in `script` option_.... so the same command works in script and not interactively...which makes it more puzzling 😅

lundmar, 2024-08-02:
I'm suspecting this may be a timing issue. What is the system you are connected in the other end? I assume some sort of modem?

lundmar, 2024-08-02:
Try remove the call to tty_sync() in line 2672 in src/tty.c - it may make a difference.

jordan-rash, 2024-08-06:
Sorry, lost track of time this weekend....i will try this and report back

jordan-rash, 2024-08-06:
Woot! that fixed it!  Thanks a ton

lundmar, 2024-08-06:
Good. I'll make the change permanent.

lundmar, 2024-08-06:
Fixed in ef12ed62df6bc5d7a2571a7a9f31be748af68cf3.

closed 2024-08-06 by lundmar