tio/tio #273: The log-directory option is not read from the configuration file.

tudorapo, 2024-08-19:
Well, what else I could say?

```
---
 src/configfile.c | 1 +
 1 file changed, 1 insertion(+)

diff --git a/src/configfile.c b/src/configfile.c
index 53545a4..97fc59f 100644
--- a/src/configfile.c
+++ b/src/configfile.c
@@ -219,6 +219,7 @@ static void config_parse_keys(GKeyFile *key_file, char *group)
     config_get_integer(key_file, group, "timestamp-timeout", &option.timestamp_timeout, 0, INT_MAX);
     config_get_bool(key_file, group, "log", &option.log);
     config_get_string(key_file, group, "log-file", &option.log_filename, NULL);
+    config_get_string(key_file, group, "log-directory", &option.log_directory, NULL);
     config_get_bool(key_file, group, "log-append", &option.log_append);
     config_get_bool(key_file, group, "log-strip", &option.log_strip);
     config_get_string(key_file, group, "map", &string, NULL);
-- 
2.43.0
```

Thanks for the otherwise nice tool!

tudorapo, 2024-08-19:
https://github.com/tio/tio/pull/274

lundmar, 2024-08-19:
Exactly. Thanks for the PR. 👍🏻

closed 2024-08-19 by lundmar