tio/tio #277: By default auto enable mapping INLCRNL when no CR with first few NL

KindDragon, 2024-09-02:
I have a lot of devices that require INLCRNL, or their output look like this:
```
Not found:
           fdsaf
                Valid commands:
                                   backend
                                                  Backend commands: 'backend <cmd_name> [cmd_arg0] ...'
                                                                                                           cpu_load
                                                                                                                           CPU load commands: 'cpu_load <cpu0_percent> <cpu1_percent>'
                 fortuna
                                ForTuna commands: 'fortuna <cmd_name> [cmd_arg0] ...'
                                                                                         lc
                                                                                                   Local automation debug commands: lc <cmd_name>
                                                                                                                                                     silabs
                                                                                                                                                                   silabs zb module commands: 'silabs <cmd_name> [cmd_arg0] ...'
                                                           sys
                                                                      system commands: 'sys <cmd_name>'
                                                                                                           telemetry
```

Maybe we could automatically detect this or at lease automatically suggests enabling this mapping?

lundmar, 2024-09-02:
I think the correct default mode for an i/o tool like tio should be no mappings as it is now.

I don't think introducing automatic detection and prompting the user is a good idea. The problem with this type of auto detection and prompting is that it may be useful for some but an interrupting annoyance for others.

Even if we did introduce such a feature it should disabled by default and in that case one might as well grab for the mapping flag instead. Either way, you will need to add configuration for your devices.

KindDragon, 2024-09-03:
Ok, but for example it work automatically with miniterm just by specifying port in miniterm
```
python -m serial.tools.miniterm /dev/cu.usbserial-0001
```

But it very hard to determine correct mapping option in TIO (or in picocom) for new user

KindDragon, 2024-09-06:
Same with picocom, it's by default use `emap is        : crcrlf,delbs,`
[image]

lundmar, 2024-09-06:
Let me think some more about this issue.

Until then you can change the default behavior of tio for all your devices by simply setting your mappings in the ```[default]``` section of your configuration file. No need to do it per device.