tio/tio #279: feat:  connect to device matched by given pattern

konosubakonoakua, 2024-09-25:
currently, I can do this.
here, cp2102 is the pattern
```bash
tio $(tio --list | grep -i 'cp2102' | grep -o '/dev/ttyUSB[0-9]*')
```
any opinion on this new feature?

lundmar, 2024-09-25:
Your command line only works if you have one cp202 device connected to your system. If you have more than one the expression will expand to multiple /dev/ttyUSB[0-9] words and that will confuse tio since it only accepts one device argument.

But I understand what you are trying to do but I think you can do it better already using the newly introduced auto connect strategies.

For example, for your case this should work:

```
tio -a latest --exclude-devices=!(/dev/ttyUSB[0-9]) --exclude-drivers=!(cp2102)
```

This will connect immediately to your latest registered cp2102 serial device.

In fact, you don't even need to specify the ttyUSB[0-9] part. Simply do:
```
$ tio -a latest --exclude-drivers=(!cp2102)
```

Notice that I'm basically inverting the exclude patters so they become inclusive only. Probably I need to explain this in the man page and provide some examples.

You can also use strategy "new" instead of "latest" to have tio wait for you to plug in your cp2102 serial device and then connect.

konosubakonoakua, 2024-09-25:
@lundmar impressive, thank you~ :D

closed 2024-09-25 by konosubakonoakua

konosubakonoakua, 2024-09-25:
BTW, there's a typo.
It should be 
```bash
tio -a latest --exclude-drivers=!(cp210x)
```