tio/tio #283: Mapping non-printable characters to a text representation (feature request)

plexonet, 2024-10-16:
When we test serial communications, sometimes the equipment sends comm data such as ENQ, ACK, STX, to enable the data transmission, it will be very useful if we can see this transmission packets in the terminal, and I know I can change to hex and see them but then I have difficult seeing the regular text packets, so a configuration to change ACK (0x06) to "<ACK>", CR (0x15) to "<CR>" and so on will help a lot, also if we send a text representation such as <ACK> and that will convert before sending to (0x06) also be a great feature.
I love this tool, btw.
Thanks.

Enrique Ortiz

lundmar, 2024-10-16:
That sounds like a useful feature to me.

I think the best way to generalize and implement this need is to add support for mappings flags that can map any character value to another value on input or output.

Maybe extend the mapping flags like so:
```
--map=OMAP(N,M) // Map N to M on output from the serial device
--map=IMAP(N,M) // Map N to M on input to the serial device
```

In your case you could then use:
```
--map=OMAP(6,-1),OMAP(21,-1),IMAP(6,-1) // Map 0x6 and 0x15 to nothing on output from the serial device 
                                        // and map 0x6 to nothing on input to the serial device
```

-1 meaning that it will do no mapping (""), simply ignore the character. Of course, instead of -1 you can use any value from 0 to 255 to map from one value to another. This would offer full flexibility to support any use case.

plexonet, 2024-10-16:
Thank you, I see where you are going to and looks good but it will have to be able to put strings in the mapping more like just numbers like so:

`
--map=OMAP(6,"<ACK>") // to map 0x6 to the stirng <ACK>
--map=OMAP(15,"<CR>") // to map 0x15 to the stirng <CR>
and
--map=IMAP(,"<ACK>",6) // this will make a conversion when I type <ACK> to 0x6 being sent instead
`
this is so we can actually see the unprintable characters being received or being able to send them if we need to.
I've noticed in my previous message that the strings I put between < and > where stripped out, that was the original intention.

lundmar, 2024-10-17:
Right, that was point wasn't clear because of the missing formatting.

I'll put this feature in the TODO list.

plexonet, 2024-10-17:
Thanks lundmar I appreciate it!