tio/tio #284: when a script is used, lua read lacks a timestamp when option.timestamp is enabled

Keith-H-Utron, 2024-10-25:
We need timestamps in the log file.  It would be helpful to add a lua function like read which also pre-pends timestamps in the system way, when option.timestamp is non-zero to its outputs, both to stdout and to the log file if enabled.  Lua function read_ts is proposed with the same meaning as read, except to add these missing timestamps.

Keith-H-Utron, 2024-10-25:
Please consider draft pull request with a prototype solution: [PR285](https://github.com/tio/tio/pull/285)

lundmar, 2024-10-25:
Thank you for prototyping the feature.

I'm pondering if maybe we should simply let read() behave as you suggest.

Keith-H-Utron, 2024-10-25:
It's not a perfect solution, because it doesn't work in a line-oriented way, but it's good enough for our needs.

lundmar, 2024-10-25:
Right, the while timestamp feature is based on lines.

The read() function is very basic and character oriented.

I'm thinking maybe we should simply go for the perfect solution and have a line oriented read_line(timeout) function which supports the time stamping feature and which features a very large internal line buffer so callers do not have to bother with the size bit in their lua scripts. We could even make timeout optional and fall back to a default timeout if none is provided. That would make yours and others scripts really simple.

Keith-H-Utron, 2024-10-25:
I like this proposal. The simpler, the better.

Keith-H-Utron, 2024-10-28:
How large should the "very" large internal line buffer be? 1k? 4k?

Keith-H-Utron, 2024-10-28:
Also, shouldn't lua read(size,waittime) also emit a timestamp as the proposed read_ts did? Because even that small improvement made the whole tio useful for our needs in the lab.

lundmar, 2024-10-28:
> Also, shouldn't lua read(size,waittime) also emit a timestamp as the proposed read_ts did? Because even that small improvement was needed before we could use tio in our data capture application. My rationale is: capturing output from stdout doubled the amount of memory needed when using pipes, which gets simplified if read simply writes the timestamps to the logfile.

Yes, it could do that.

lundmar, 2024-10-28:
> How large should the "very large" internal line buffer be? 1k? 4k?

I would say on the very large side, 16k or more. Simply make it a static char array so it gets allocated from heap.

Keith-H-Utron, 2024-10-30:
please consider [pull request 286](https://github.com/tio/tio/pull/286) as a solution

Keith-H-Utron, 2024-11-07:
Was merged to master. Thanks for the guidance, Martin.

closed 2024-11-07 by Keith-H-Utron

lundmar, 2024-11-07:
> Was merged to master. Thanks for the guidance, Martin.

No problem. Thank you for your contribution!