tio/tio #292: Simple script not working as expected

viperk1, 2025-01-26:
Hey folks, first of all, this is a great tool and thank you for all your work.

I'm trying to script some commands to a switch connected over serial console. Maybe my understanding of the scripting is flawed but I thought the commands were sent locally over the opened terminal session. However, every time I try to run a script it tries to send it over xmodem, which fails. Is there something I'm doing wrong? I tried it with an extremely basic command to start (send a newline).

```
tio --baudrate 9600 --script 'send("\n")' /dev/aruba-backup
[10:19:25.043] tio v3.7-16-g01e637c
[10:19:25.044] Press ctrl-t q to quit
[10:19:25.047] Connected to /dev/aruba-backup
[10:19:25.049] Running script
[10:19:25.049] Sending file '
                             ' using XMODEM-1K
[10:19:25.049] Error: Could not open file
[10:19:25.049] Aborted
[10:19:27.331] Disconnected

```

Thanks in advance!

lundmar, 2025-01-26:
> Hey folks, first of all, this is a great tool and thank you for all your work.

Thanks! Glad you like it.

> tio --baudrate 9600 --script 'send("\n")' /dev/aruba-backup

In recent versions of tio send() is for sending files while write() is for writing strings.

viperk1, 2025-01-26:
Prompt response and immediate solution. Thank you so much!!!

```
tio --baudrate 9600 --script 'sleep(1); write("\r\n");' /dev/aruba-backup
[10:49:34.131] tio v3.7-16-g01e637c
[10:49:34.132] Press ctrl-t q to quit
[10:49:34.136] Connected to /dev/aruba-backup
[10:49:34.138] Running script
[10:49:34.139] Sleeping 1 seconds

User:
```

closed 2025-01-26 by lundmar