tio/tio #296: Input ICRCRNL mapping

ekawahyu, 2025-03-06:
Some of my devices require ICRCRNL mapping, which does not exist in tio, but picocom does support it by passing `--imap crcrlf`

Example on picocom:

```
picocom -b 115200 /dev/tty.usbserial --imap crcrlf
```

It can be done with tio 3.8. It kind of works, but it does not feel intuitive at all:

```
tio -b 115200 /dev/tty.usbserial -m INLCRNL -m ICRNL

  or

tio -b 115200 /dev/tty.usbserial  -m ICRNL -m INLCRN
```

Here is a list of picocom CR and NL mapping, that works for both input AND output, depending on whether you pass it with `--imap` or `--omap` prefix:

```
  crlf : map CR --> LF
  crcrlf : map CR --> CR + LF
  igncr : ignore CR
  lfcr : map LF --> CR
  lfcrlf : map LF --> CR + LF
  ignlf : ignore LF
```

Unless I am missing anything in the documentation about how to use input mapping, this will be a feature request then.

lundmar, 2025-03-06:
Yes, this is indeed a feature request. Will put this on the TODO.

dordnung, 2025-04-22:
I've submitted a pull request implementing this feature, as I also needed it to log the actual input rather than the manipulated one: https://github.com/tio/tio/pull/311

lundmar, 2025-04-23:
@dordnung Thank you for the PR fixing this issue. It is much appreciated since I'm currently swamped with work so I don't have much time to solve issues myself.

closed 2025-11-12 by ekawahyu