tio/tio #302: Can't get a script to work

munnik, 2025-03-13:
Hi there,

`tio` looks like the perfect tool to manage some serial to USB converters we use in the field. These devices can be managed via a serial port, and so far we did that manually using `screen`. Because we have more and more of these devices, we want to automate the process. I just can't get it working, it seems the commands I specify don't arrive at the serial device. When I use `tio` and enter the commands manually, it works fine.

A manual attempt:
```
$ tio -b 38400 /dev/ttyUSBComManagement
[06:58:07.747] tio v3.7
[06:58:07.748] Press ctrl-t q to quit
[06:58:07.748] Connected to /dev/ttyUSBComManagement
[8#1009] USB-COM Plus

 0 - Exit
----------------------------------
 1 - Show all port configurations
 2 - Change port configurations
 3 - Default port configurations
 4 - Save port configurations
----------------------------------
 5 - Show configuration file
 6 - Upload configuration file
> 1

DIP Switch: by Software (none)

Software Configuration:

Port 1: RS232
Port 2: RS232
Port 3: RS232
Port 4: RS232
Port 5: RS232
Port 6: RS232
Port 7: RS232
Port 8: RS232

 0 - Exit
----------------------------------
 1 - Show all port configurations
 2 - Change port configurations
 3 - Default port configurations
 4 - Save port configurations
----------------------------------
 5 - Show configuration file
 6 - Upload configuration file
>
```
The prompt is provided by the serial device is `>`. In this example, I pressed <kbd>↵</kbd> after connecting. This will show the main menu, then I pressed <kbd>1</kbd> and <kbd>↵</kbd>. Now I want to achieve the same with a script, so I run:
```
$ tio -b 38400 /dev/ttyUSBComManagement --script 'io.write("\n");expect(">");io.write("1\n");expect(">");'
[07:10:41.815] tio v3.7
[07:10:41.815] Press ctrl-t q to quit
[07:10:41.816] Connected to /dev/ttyUSBComManagement
[07:10:41.816] Running script

[07:11:07.059] Disconnected
```
The script keeps running forever, and I have to quit with <kbd>Ctrl+t</kbd> <kbd>q</kbd>. I also tried:
```
$ tio -b 38400 /dev/ttyUSBComManagement --script 'io.write("\n");expect(">");io.write("1\n");expect(">");' --map INLCRNL,OCRNL
[07:11:30.242] tio v3.7
[07:11:30.242] Press ctrl-t q to quit
[07:11:30.243] Connected to /dev/ttyUSBComManagement
[07:11:30.243] Running script

[07:11:32.537] Disconnected
```
with exactly the same results. What am I doing wrong or how can I troubleshoot/debug this issue?

lundmar, 2025-03-13:
> `tio` looks like the perfect tool to manage some serial to USB converters we use in the field. These devices can be managed via a serial port, and so far we did that manually using `screen`.

I hope so. This is why I introduced the new script feature :)

The problem is your use of io.write(). You need to use the tio lua API for writing to the serial device.

See https://raw.githubusercontent.com/tio/tio/refs/heads/master/man/tio.1.txt in the "SCRIPT API" section.

munnik, 2025-03-13:
tio v3.8 solved the issue!

closed 2025-03-13 by munnik

lundmar, 2025-03-13:
> tio v3.8 solved the issue!

Hehe. Yes, I'm sorry, you are hitting a cutting edge feature here so expect bugs. Feel free to let me know any issues or quirks you run into regarding your specific use case. We may want to fine tune this feature a bit.

lundmar, 2025-03-13:
Oh, and by the way, there are some more elaborate examples e.g. here: https://github.com/tio/tio/blob/master/examples/lua/automatic-linux-login.lua