tio/tio #303: Request for Precompiled Binaries in GitHub Releases for Mainstream Architectures

rainote2020, 2025-03-13:
First, thank you for developing tio — it’s truly the most modern and user-friendly serial tool I’ve encountered,

However, there’s a significant usability gap for Linux users:
- Ubuntu/Debian repositories ship an extremely outdated version (e.g., 1.32 in Ubuntu 22), lacking many features.
```
Package: tio
Architecture: arm64
Version: 1.32-1
Priority: optional
Section: universe/comm
Origin: Ubuntu
```
- While compiling from source works, dependencies like meson and ninja add friction for non-developers or those in restricted environments.

**Suggestion:**
Could precompiled binaries for mainstream architectures (x86_64, ARM64) be added to GitHub Releases? Many CLI tools provide static binaries, greatly simplifying installation for end-users.

**This would:**
Make cutting-edge tio features accessible without compilation.
Reduce dependency hurdles (especially on minimal/embedded systems).
Encourage adoption across platforms like Raspberry Pi (ARM) and standard Linux workstations.

Thank you again for your work on this excellent tool!

lundmar, 2025-03-13:
> First, thank you for developing tio — it’s truly the most modern and user-friendly serial tool I’ve encountered,

Thank you very much for the kind praise ❤ . I'm glad you like the tool. I believe it is one of the best serial tools available because of all the contributors work.

> However, there’s a significant usability gap for Linux users:
> 
>     * Ubuntu/Debian repositories ship an extremely outdated version (e.g., 1.32 in Ubuntu 22), lacking many features.

True, this is a problem caused by the nature of how projects evolve. It is what it is.

> * While compiling from source works, dependencies like meson and ninja add friction for non-developers or those in restricted environments.
>
>
> **Suggestion:** Could precompiled binaries for [...]

It is possible. This means setting up meson so it has a build target which compiles tio and its dependencies (glib and lua) statically. Then build on the most popular architectures (x86 and arm, 32bit vs 64 bit) in e.g. github CI pipelines so it is triggered automatically on release.

We can add it to the growing list of TODO things. Unfortunately, At the moment I'm very busy with IRL freelance projects so I don't have the time to look into such setup anytime soon.

If you want this feature sooner rather than later I encourage you to give it a try yourself.

> Thank you again for your work on this excellent tool!

You're welcome and thanks to all the contributors.

sur5r, 2025-03-13:
Debian Maintainer of tio here.

Ubuntu 22.04 was released in, well, 2022. 1.32 was the current version of tio back then. If you are using an old distribution, please don't complain upstream that your software is outdated. This is not a usability gap, it works as intended. If you want a newer version of tio for 22.04, please file a bug with Ubuntu so they can consider a backport.

Regarding Debian: 
```
$ rmadison tio
tio        | 1.32-1        | oldoldstable             | source, amd64, arm64, armhf, i386
tio        | 1.32-1        | oldstable                | source, amd64, arm64, armel, armhf, i386, mips64el, mipsel, ppc64el, s390x
tio        | 2.5-1~bpo11+1 | bullseye-backports       | source, amd64, arm64, armel, armhf, i386, mips64el, mipsel, ppc64el, s390x
tio        | 2.5-1~bpo11+1 | bullseye-backports-debug | source
tio        | 2.5-1         | stable                   | source, amd64, arm64, armel, armhf, i386, mips64el, mipsel, ppc64el, s390x
tio        | 3.7-1~bpo12+1 | stable-backports         | source, amd64, arm64, armel, armhf, i386, mips64el, mipsel, ppc64el, s390x
tio        | 3.7-1~bpo12+1 | stable-backports-debug   | source
tio        | 3.8-1         | testing                  | source, amd64, arm64, armel, armhf, i386, mips64el, ppc64el, riscv64, s390x
tio        | 3.8-1         | unstable                 | source, amd64, arm64, armel, armhf, i386, mips64el, ppc64el, riscv64, s390x
tio        | 3.8-1         | unstable-debug           | source
```

You can get 3.7 for stable right now and a backport of 3.8 will be added soon. So please don't tell me that tio on Debian is *extremely outdated* as that is just not true.