tio/tio #304: Hardware flow control ineffective?

ceharris, 2025-03-18:
It seems the `--flow hard` option to enable RTS/CTS flow control is ineffective when sending a large volume of text (e.g. pasting text from a file, or running a shell command to `cat` the contents of a file) -- characters are lost at the receiver even though the receiving device is de-asserting RTS to inform the sender that it it should pause. 

Using the same receiving device, same USB serial cable, and same serial port settings (115200,N,8,1) with hardware flow control enabled in `minicom`, the flow control works as expected -- when either pasting a large body of text or using the option to send an ordinary text file, no characters are lost.

Perhaps related:
Adding command line options such as `--output-delay` or `--output-line-delay` has no effect on this behavior. With a long enough delay, the effect of these options can easily be observed when typing at the terminal. However, when pasting text or running a shell command that produces text output, there's no apparent change in the pace at which text is being delivered to the receiver.

Additional context:
I'm trying to use `tio` in a retro-computing environment in which the workflow involves sending Motorola S-Records to a receiving device over the serial line, and RTS/CTS flow control is used to avoid overrunning the receiver's buffer.

lundmar, 2025-03-22:
@ceharris Interesting. As far as I can tell we are setting the correct bits in the tio code to enable hardware flow control so I'm a bit puzzled why it should not work. I mean, we may have several layers of buffering going on which can affect timing but still the control lines RTS/CTS should be respected on the hardware level.

ceharris, 2025-04-29:
I'll try the debugging actions that you mentioned and report what I learn.

ceharris, 2025-05-02:
I confirmed that the `stty -a` output for the device is `-crtscts` by default and is toggled to `crtscts` when `tio` is invoked with `--flow=hard`. Same for `minicom`.

This weekend I'll hook up the scope again and get some traces that show the timing of the remote's de-asserting RTS (CTS at the local) and the pause (or lack thereof) in transmission towards the remote.

The mention above of multiple layers of buffering may indeed be a factor. As is typical of these old 8-bit systems, the receive buffer is only 256 bytes, and without looking at the code, I suspect that it de-asserts RTS/CTS at a low water mark of something like 16 or 32 bytes. Perhaps there's some difference in buffering strategy between `tio` and `minicom` that's significant in how quickly a change in CTS can pause transmission.

lundmar, 2025-05-02:
This is an interesting discussion. My gut feeling is that this is a timing issue induced by the internal I/O path of tio. When tio reads data from input it does so pushing data through a fifo pipe between an input thread and the main thread. The input thread was created because tio is using forever blocking read/write operations and the input thread simply helps solve the following problem as described in commit 93e6efc00153e625e4ceb8a51e01b324247d312a : 

"To make tio more responsive to quit and I/O flush key command when main I/O
thread is blocked on output."

It ended up being more complex that I wanted and today I think I would make a simpler more direct I/O path with one poll/select mechanism basing I/O on timeout so there will be room to handle e.g. quit and flush commands from the user between timeout periods.

Add to that the internal buffering mechanics which may result in tio having less performant I/O performance. There have been some reports of e.g. x-modem file transfers being less performant compared to other tools. In short, I know the I/O path in tio isn't optimal but I haven't had time to rework it yet.

lundmar, 2025-05-03:
> CU runs two processes in userspace for reasons similar to yours.
> Non-blocking reads and writes are certainly possible, but they're just
> plain harder.

Right. Maybe the current thread model is not a bad choice. Still could use a review though. Maybe it is the buffering/caching mechanism that is the culprit here.

> In case I need credentials, I was a Sr. Engineer for the world largest
> maker of TTY system. I don't know the specifics of Linux, but I've been
> around the block with Unix internals for almost 40n years. So maybe I can
> do "interesting" 👴.

That is pretty cool! Clearly you know TTY systems and now we know why :)

If you are looking to involve yourself in an open source and TTY related project then tio may be a perfect fit. tio is a fairly popular alternative to some of your older classic tty applications. However, it has grown organically to a point where tio needs more maintainers to move forward so tio can continue improve and add new features relevant for developers and hackers working with serial devices. If you are interested I would be glad to offer such role :)

lundmar, 2025-05-03:
>  looking to involve yourself [ ... ]  tio may be a perfect fit.
> LOL. You know I've been here off and on for almost a year now, right? I
> left a standing offer for intermittent help. (I'm too unreliable and flaky
> to be counted on...)
> I'll leave my employment history after Digi a secret mystery. :-) I've been
> involved in open source since the 80s, with my hands in probably hundreds
> of apps by now.

Of course, but I just had to put the co-maintainer idea out there. I'm hitting the limits of what I can do for the project as one maintainer :D

ceharris, 2025-05-10:
After dorking around a bit with the scope trying to get a sense of the "skid" length when the remote de-asserts CTS and not having much luck, I decided to write a couple of small C programs to act as sender and receiver and connect two serial ports back to back on the same host, with a null modem cable. The sender sets up its port for hardware flow control and waits for a key press before it begins transmitting a sequence of printable ASCII characters. The receiver also sets up its port for hardware flow control, then it enters a loop where it reads 100 characters, then de-asserts the sender's CTS, then counts the number of characters received before the next read would block -- i.e. it measures the skid length. It then repeats, reading another 100 characters, and again measuring the skid length. It computes the min, max, and mean skid lengths for 1000 passes.

Min skid length is consistently 6 characters. Mean is 8 characters. At the other end there tend to be a few outliers -- skid lengths that are more than 64 characters. Out of 1000 passes, typically there just one or two outliers over 64 characters, and none over 80 characters. So if a slow receiver wants to minimize the potential for lost characters, it needs to de-assert the sender's CTS with enough remaining buffer capacity for 80-ish characters.

With that first experiment working, I decided to use `tio` as the sender and repeat the experiment, just to see if I would get similar results. To make this work, I created a large text file with a sequence of printable ASCII characters and used the option to redirect output from a command towards the receiver. So I could characters to my receiver using `cat text.dat` and do the same skid length analysis in the receiver. What I learned was very surprising.

For approximately 40 passes, the receiver counted skid lengths comparable to what I'd seen in the first experiment. Most importantly *the flow control is indeed working* with `tio` as the sender. But after ~40 passes, the receiver basically blocked waiting for more characters -- CTS was asserted towards `tio`, but `tio` sent nothing more.

I didn't notice what was happening until I tried it a couple of times. After observing the same behavior on a second attempt, I noticed something I hadn't noticed before -- the `cat` command was terminated prematurely:

```
[13:30:22.483] Execute shell command with I/O redirected to device
[13:30:27.652] Executing shell command 'cat text.dat'
[13:30:27.654] Command exited with status 1
```

I think what may be happening is that when the sender's buffer fills and the next write would block, the `cat` command is exiting with an error. The fact that it happens about 40 passes through my test loops is consistent with that -- some other testing I did suggests that the OS has a 4 KiB buffer on the transmit side for the serial device; 40 passes of 100 characters, plus skids is certainly in the right ballpark. If the descriptor that was passed to `cat` as standard output is set up for non-blocking mode, the first write when the buffer is full due to flow control would return error EWOULDBLOCK.

So bottom line, I think the flow control is working correctly, but there's something going wrong that prevents the whole file from being transmitted.

If `tio` had a option to simply transmit a file to the remote by simply reading the file and sending it with no other protocol mechanism, it would probably work just fine.

lundmar, 2025-05-10:
@ceharris Thanks for the deep dive. Instead of using the --exec feature, can you repeat your test using e.g. ```cat text.dat | tio /dev/ttyUSB0``` and see if the same happens. Thanks.

ceharris, 2025-05-10:
Similar result, I think:
```
ceharris@kahoolawe:~$ cat text.dat | tio --flow=hard /dev/ttyUSB0
[15:34:38.924] tio v3.9
[15:34:38.924] Non-interactive mode enabled
[15:34:38.924] Press ctrl-c to quit
[15:34:38.926] Connected to /dev/ttyUSB0
[15:34:38.935] Disconnected
[15:34:39.334] Error: Could not write to serial device (Resource temporarily unavailable)
```

ceharris, 2025-05-10:
I think that seals the deal on it being a side effect of using flow control when the device is configured with O_NONBLOCK. The write is getting error EAGAIN, and should be retried.

lundmar, 2025-05-10:
Right, good catch. I see the non-interactive piping situation is a bit different i/o wise and it should of course handle EAGAIN and EWOULDBLOCK or maybe simply reconfigure it blocking.

lundmar, 2025-05-11:
@ceharris There is a branch that fixes the piping situation here: https://github.com/tio/tio/tree/non-interactive-write-fix Feel free to give it a test.

ceharris, 2025-05-11:
The good news is that the fix (with a slight correction) prevents it from failing due to EAGAIN. The bad news (for me) is that with that fix, now the skid lengths are averaging a few hundred characters, which is way too much for my remote to handle. Not sure why the fix changes that aspect so dramatically.

The slight correction I mentioned above is that I think you meant to turn off non-blocking I/O, which means that you want this instead of what's in the commit on the branch you mentioned.

```c
fcntl(device_fd, F_SETFL, flags & ~O_NONBLOCK)
```

lundmar, 2025-05-11:
> The slight correction I mentioned above is that I think you meant to turn off non-blocking I/O, which means that you want this instead of what's in the commit on the branch you mentioned.
> 
> fcntl(device_fd, F_SETFL, flags & ~O_NONBLOCK)

Fixed. I was too quick last night and pasted the wrong buffer.