tio/tio #306: Tio on MSYS2 connects then immediately disconnects.... UNLESS you run PuTTY first(?!?!)

b4shful, 2025-03-30:
Apologies for the far from comprehensive write-up here. I'm normally using Linux but had to use Windows for a few weird embedded dev things. Hence, I've been using tio on Windows via MSYS2. After experiencing this issue for the past week since I downloaded tio on MSYS2, I decided to go down this rabbit hole and after getting suitably stuck, I'm now hastily writing this so that I can leave it for now and get back to what I was doing 😆 

When I try to use tio, I get a constant loop of connected then immediately disconnected, followed by the reconnection a second or so later which also immediately disconnects. I didn't get any useful errors until I realised that this is probably just tio's reconnect loop, and so when passing `--no-reconnect` I got:

```
$ tio -n COM4
[06:08:58.902] tio 3.8
[06:08:58.903] Press ctrl-t q to quit
[06:08:58.904] Connected to /dev/ttyS3
[06:08:58.905] Disconnected
[06:08:58.906] Error: Could not get port settings (Invalid argument)
```

However, and I have no idea what sorcery is taking place here, but when I open another Windows serial terminal app, like PuTTY, on the same COM port, with the same settings, PuTTY stays connected just fine. Then, if I close PuTTY's session/close PuTTY completely, tio magically stays connected!
(I removed `-n` for this one, no longer necessary here as there's no errors, it stays connected with no issues)
```
$ tio COM4
[06:12:59.689] tio 3.8
[06:12:59.690] Press ctrl-t q to quit
[06:12:59.691] Connected to /dev/ttyS3
<and it works!>
<pretend there's some output here - I didn't tell my embedded device to output anything on the com port>
```

I've tried MSYS2 UCRT64 and MINGW64 with the tio MSYS2 package, as well as building tio myself in MSYS2 (using a process based loosely off of the tio MSYS2 package's PKGBUILD), and I could replicate this issue in all of them.

If you want me to try anything and provide further info do let me know. I also have a debug build of tio I can use to gather more info if needed (well, Meson's debug build type with the DEBUG macro defined, which I added after seeing `tio_debug_printf` was a thing, but then I later realised it's not called in very many places so it didn't give me any further insight). So, if you would like me to poke a debugger at anything to get some more info I can also do that, although perhaps that may not be all that useful for this issue.

lundmar, 2025-03-30:
@b4shful Thanks for reporting the issue. This sounds like tio does not get the serial port initialized correctly but somehow Putty does and then tio can work on top of that. It has been a long time since any tio msys2 issues have been reported. Perhaps something changed in Windows. What version of Windows are you using? Very latest?

b4shful, 2025-03-30:
Oh yeah I forgot to add that part. I'm on Windows 11 24H2 (26100.3476).

I'm not sure how long Putty's "fix" works for. But when I do it, I close putty's session fully and close the putty application fully. Whether it stays good until the MSYS2 shell is closed and re-opened, or if it stays good until the USB Serial device is unplugged/replugged, I'm not quite sure. At some point later today I'll try to find out the answers to that/find out what exactly makes it go back to the broken state.

Just as a note, my Putty version is 0.83 and was installed from the usual MSI installer available on their site. Their installer builds for Windows seem to use the following:
```
Build platform: 64-bit x86 Windows
Compiler: clang 14.0.6 , emulating Visual Studio 2022 (17.2), _MSC_VER=1932, _MSC_FULL_VER=193231329
```
I guess since it's not an MSYS2 build of putty and is about as native as it gets (MSC and such), I haven't confirmed there's a chance if something changed in Windows, the problem needing fixed might lie somewhere else upstream (pure speculation though - I don't know much about MSYS2)

lundmar, 2025-03-30:
True, it could be a MSYS2 regression. I just checked the tio code, the only code bits that does things differently on windows builds is https://github.com/tio/tio/blob/d33e275ca3ba5c40435fcbf6c6fb9562c073e498/src/options.c#L1162 where we translate COMx to /dev/ttyS(x-1) and from your output that seems to be working fine. Everything else should just be standard Linux port handling up against the MSYS2/MINGW APIs.

lundmar, 2025-03-30:
Also, what kind of serial port hardware are you using? Maybe some special kind of usb-serial adapter?

b4shful, 2025-03-30:
Oh dear...

I've been trying serial devices. It's all USB serial to some extent, but via various means.

I grabbed an ESP32-S3 (ESP32-S3-DevKitC-1) I had lying around and flashed blinky onto it (which logs out to the serial port every time the LED blinks on and off) and *tio worked first time*.

The device I've been working on and encountered the problem with is an STM32 (Nucleo) H755ZI-Q devkit. After the ESP worked, I stopped the program, unplugged it, plugged in the STM32 and found its COM port (assigned different from the ESP's, so I think Windows's COM port enumerator is working fine), then tried to connect to it and got the usual `Error: Could not get port settings (Invalid argument)`.

Running out of serial port devices, I tried my Flipper Zero next, which has a USB to UART Bridge app that can be run (https://docs.flipper.net/gpio-and-modules). The Flipper shows up as a COM port when connected to the computer via USB (it has Flipper CLI, a terminal interface which can be accessed through the USB Serial Port. When you open the USB to UART Bridge app, you can have it send/receive UART through the CLI, or you can change the USB Channel setting from 0 (CLI) to 1, which causes the Flipper to disconnect briefly before it reconnects showing up as two USB Serial Devices. The new USB Serial Device is a dedicated COM port for the UART bridge. In my case, it showed up as COM8 (the CLI) and COM9 (UART bridge). These all just use Microsoft's `usbser` driver.

When I tried that, both the CLI and UART Bridge COM ports gave me the same error I've been getting so far (`Error: Could not get port settings (Invalid argument)`). If I forget about the UART bridge functionality and just simply try to access Flipper's CLI through the single COM port it shows up as by default, I also get the same error.

And in this case, the putty thing once again works. Opening putty:
<details>
<summary>Opening putty... (click to show image)</summary>

I get the expected output: 

[image]

</details>

<details>
<summary>Then close putty and go back to tio (click to show image)</summary>
Now it works!

[image]

</details>

Flipper Zero is an STM32 based device too (uh oh..). I'd really hope it's not some random quirk of the STM32 causing the issue. I tried the flipper because unlike my STM32 code, I can trust that they're doing everything correctly when it comes to USB CDC/Serial. I'll update this post after I try some other serial adapters (I think I have some FTDI chip based ones somewhere). Or maybe I should try to set up some virtual COM ports (com0com/hub4com style) and see if I can replicate the problem with those as well.

b4shful, 2025-05-13:
Small update - I *think* this issue only occurs when using Tio with an STM32 device that uses STMicroelectronics USB Device Middleware.

I recently moved away from the ST USB stack to TinyUSB and, while I haven't 100% verified that the issue is completely gone (as in, haven't rebooted and such to see if it randomly comes back), it would definitely appear that it doesn't happen when using TinyUSB.

I wonder if it's something to do with ST's line coding or some quirk of their implementation?

Pinging @HiFiPhile here in case you happen to know anything about this. (I was using your TinyUSB BSP PR and cache fixes for NUCLEO-H7S3L8 which I happen to be working with a lot now, and then noticed you also wrote a "hacky win32 port" for Tio recently - I was thinking "are we the same person?" lol). If you happen to know what might be going on, it would be appreciated, but no worries if you don't wish to respond :)

lundmar, 2025-05-13:
> I wonder if it's something to do with ST's line coding or some quirk of their implementation?

That is possible - perhaps the ST USB CDC feature is simply unstable so the device disconnects/connects repeatedly.

HiFiPhile, 2025-05-14:
> Small update - I think this issue only occurs when using Tio with an STM32 device that uses STMicroelectronics USB Device Middleware.
I recently moved away from the ST USB stack to TinyUSB and, while I haven't 100% verified that the issue is completely gone (as in, haven't rebooted and such to see if it randomly comes back), it would definitely appear that it doesn't happen when using TinyUSB.

I haven't touched ST's USB stack for a long time which is PITA, that's how I started to work on TUSB.

MSYS2's serial translation layer is definitely buggy, not only has timing issue but line state is not working correctly #230 #233. 
To be short on two PC with the same Windows 10  22H2 build, same MingW packages version, same CP2102 dongle, same tio v2.7 and same driver, one reads/writes line state correctly but another not...

To workaround this I had to built tio natively which works well.

lundmar, 2025-05-14:
@HiFiPhile Interesting. Thanks for letting us know.