tio/tio #308: Not sure how to use pattern to match dev/tty-port-right-upper, tio coredumping

cnf, 2025-03-31:
I am on a framework, and I have udev rules to make tty device names depening on what port they are plugged in, so i en up with

- /dev/tty-port-right-upper
- /dev/tty-port-right-lower
- /dev/tty-port-left-upper
- /dev/tty-port-left-lower

I am trying to match so I can do `tio tty-right-upper` for example, but i'm failing to figure out.

```config   
[usb-ports]
pattern = ^tty-(left|right)-(lower|upper)
device = /dev/tty-port-%m1-%m2
local-echo = true
map = INLCRNL,ODELBS
color = 165
```

With the above config, I get the following output:
```
tio tty-right-upper
corrupted size vs. prev_size
                            zsh: IOT instruction (core dumped)  tio tty-right-upper
                                                                                   %
```

Everything else I try doesn't match on the pattern.

Any hints what I am doing wrong, or did I run into an actual bug?

Thanks!

lundmar, 2025-03-31:
@cnf I can confirm that it is a real bug. You are making the right setup because if you change your device string to /dev/tty-port%m1%m2 you will see m1 and m2 be correctly matched and replaced with the right strings. However, /dev/tty-port-%m1-%m2 somehow makes it crash. I don't have time to dig deeper right now to fix the issue but I will put it on the todo.

cnf, 2025-03-31:
great, thanks for confirming. I can privately share coredumps if you need.

cnf, 2025-04-02:
I don't know if this helps a lot, you can do the same thing easy, but this is with debug compiled:

```sh
$ tio tty-right-upper
[debug] Full match: tty-right-upper
[debug] Subexpression 1 match: right
[debug] Subexpression 2 match: upper
corrupted size vs. prev_size
                            Aborted (core dumped)

```

lundmar, 2025-04-13:
Finally got around to debugging and fixing the issue. Fixed in f87f470415ad4e257f8ed8d8e2673baf5b9dbecb and included in the just now released tio v3.9

closed 2025-04-13 by lundmar

cnf, 2025-04-15:
Hi, thanks for this! I haven't tested it yet, as I'm trying to do 10 things at the same time, and I am on nixos, and i am failing to build tio right now, but i'll get back to it soon, really appreciate your work!

cnf, 2025-04-15:
figured it out, was me doing a dumb :P tested, and confirmed working, thanks!