tio/tio #309: Guided connection

knivd, 2025-04-01:
This is not an issue per se but more like a feature request.
It would be great if there is an option to get tio to list the serial ports and let user choose one for connection in a interactive way on start. Basically, replacing the manual sequence:
tio -l
<copy port name from the list>
tio -b xxxx <paste the port name>

lundmar, 2025-04-05:
I see your point but I'm undecided. I'm not sure how useful this will be for most users. I mean, beginners will likely not be aware of such option, and I don't think power users will ever use it, they will rely on auto completion.

Also, we have interactive options during sessions and I'm not sure going down the path of introducing interactivity on the command line options is a good idea.

lundmar, 2025-04-24:
> Unfortunately, on my macOS system, it just makes a jumble of the screen and doesn't appear to connect in either case. I thought I could dial it in with --exclude (and a stty sane to the controlling tty so that OPOST worked) to narrow the pool that --list or --auto would be looking at, but that doesn't appear to limit the scope. But let's not get distracted by what doesn't work for me.

If I recall correctly there may be an issue with how mac os reports device age/birth/creation time on which the connect strategies are based on. This may be why it does not work on mac os.

lundmar, 2025-04-24:
> Let's maybe fork that to another conversation. Can you confirm that --auto
> should work as I described above? If so, that probably should solve the
> problem for the OP, I think.
> […](#)

Right. Yes, it should work as described.

lundmar, 2025-04-25:
> Does it work as described ANYWHERE, and if so,. Where?  I think that if it
> worked as described, that would solve  the original problem, so OP should
> be motivated to help us check it out.

Right, let me qualify my answer by saying that the auto connect feature works as described on native Linux platforms.

> I'll help investigate, but it'll be later in the weekend. Knowing if it's a
> Mac thing or not would help reduce the blast radius to chase.

In your time. Any help is appreciated since I don't have the setup to test on other platforms but Linux.

lundmar, 2025-04-25:
> No problem. I may have some relevant skills here. Maybe we can help [@knivd](https://github.com/knivd)
> in the process, but selfishly, those sound like pretty nifty features -
> which I hadn't even noticed until yesterday. :-)
> […](#)

Right. Yes, I think that with the added auto connect strategies, tio covers what most users need in terms of connect features. I do think they are pretty nifty features, especially the connect by topology ID but I realize this is probably the least used feature but great for those that need to connect to the same usb serial device which does not have a normal serial-by-id appearance. This only works on Linux because it is based on the kernel exposing its device topology tree, which I'm not sure is available on mac.

knivd, 2025-04-25:
Thank you all for the comments. Unfortunately, the -a option is not applicable in my case since I am talking about an embedded Linux system in which the ports have been pre-initialised during startup. What I was asking about is literally the equivalent of
tio -l
then tio -b xxxxx followed by the device name. Unfortunately, those are quite long in linux. I agree that an external script can do the job, just thought it would be great if there was no need for such in first place

jcormier, 2025-06-12:
> they will rely on auto completion

Huh wow, I never thought to try and auto complete.  Thats really helpful. Can I suggest adding that to the list of Features on the README, a helpful addition on top of auto connect.