tio/tio #313: Forward terminal information

Joery-M, 2025-04-24:
When connected to a target with tio, i would like to have terminal information forwarded like color, columns and rows.

Currently, that doesn't happen:
```
[root@computer ~]# echo $COLUMNS $LINES $TERM
209 52 xterm-256color
```
```
[root@computer ~]# tio /dev/ttyACM0
[14:15:55.838] tio v3.9
[14:15:55.838] Press ctrl-t q to quit
[14:15:55.839] Connected to /dev/ttyACM0
root@remote:~$ echo $COLUMNS $LINES $TERM
80 24 vt220
```

This would also require listening to SIGWINCH for terminal size changes

lundmar, 2025-04-28:
I agree, this is out of scope of tio.

Joery-M, 2025-04-29:
Understandable.

Then for future reference, heres a Lua script i wrote with a bit of googling and 0 understanding of the actual language:
```lua
function trim(s)
    return (s:gsub("^%s*(.-)%s*$", "%1"))
end

local function exec(cmd)
    local handle = io.popen(cmd)
    local result = handle:read("*a")
    handle:close()
    return trim(result)
end

cols = exec("tput cols")
rows = exec("tput lines")
term = os.getenv("TERM")

write("\nstty cols " .. cols .. " rows " .. rows .. " && export TERM=\"" .. term .. "\" && . ~/.bashrc\n")
```

closed 2025-04-29 by Joery-M