tio/tio #319: [request] script aliases

thoughtpatterns, 2025-05-18:
hey! ive recently found tio and love it, its a step up from the simple qemu console :) i have a couple of scripts i run semi-frequently (ex. to resize the console) and was curious as to whether a way to alias scripts to a name (maybe with a different hotkey) would be appreciated. i wouldnt mind trying it myself but figured id ask first to see if itd be welcome. (it may be cleaner to add the ability to map key sequences, ex. let ctrl-t u = ctrl-t r /path/to/script, but that might be a bit out of scope for this feature.)

lundmar, 2025-05-19:
@thoughtpatterns Sure, good ideas are welcome and this sounds like a useful feature. The question is how to implement the user interface to make it easy to use.

yabu76, 2026-02-14:
Hi all, 
I tried to implement this feature in https://github.com/yabu76/tio/tree/luaapi_twrite.

It enables mapping a key to lua-script-file or lua-function, not to another key-mapping.
I think it will become even more useful as more LuaAPIs are added. :-)

2026-02-08 16:17:30
    Add user key-script mapping function (--keymap, Ctrl-t k)
    
    User key-script mapping function:
    
    You can specify the mappings as @<key-1>=<script-description-1>
    @<key-2>=<script-description-2>... @<key-N>=<script-description-N>.
    
    Script-description is script-filename or '!'script-commands.
    
    After that,
    When you press ctrl-t and <key-n>, tio executes <script-description-n>.
    This user keymap takes precedence over the default settings (except for
    ctrl-t q).
    
    Example of startup option:
    tio /dev/ttyUSB1 --keymap '@1=!print(tio.banner())
    @2=!tio.write("test\r") @ctrl-a=!tio.write("ctrl-a\r")
    @ctrl-j=test-script.tio'

    Example of .tioconfig: (note: backslash escape needed.)
    keymap = @1=!print(tio.banner()) @2=!tio.write("test\\r")
    @ctrl-a=!tio.write("ctrl-a\\r") @ctrl-j=test-script.tio

Have a nice weekend :-)