tio/tio #321: Thank you for this project!

noahp, 2025-05-23:
Just wanted to say thank you for creating `tio`, I use it constantly! Superbly useful tool 😍 !

lundmar, 2025-05-23:
@noahp I appreciate you saying that and I'm sure all the contributors do too! Thanks!

jcormier, 2025-06-12:
Agreed.  I've used screen/putty/teraterm/minicom over the years.  Tio has replaced all for my daily usage.

> what do you think are the best features of tio that you use on the daily

Daily specifically:
* Connect to last/next
* List connected devices
* Ability to config specific device names so I don't have to search dmesg for the correct serial port every day
* Reasonable defaults for baud rate (thank you embedded vendors for all deciding on 115200 8n1)
* Easy of logging (Generate unique name and timestamped)

Stuff I'm excited to try:
* Timestamping incoming text (Teraterm supports this but I haven't found a simple way of doing this in Linux)
* LUA scripting (Doing something as simple as waiting for boot, logging in, and rebooting the device required a bit too much python and i'd lose the ability to interact with the device)

FerriteGiant, 2025-07-25:
Just to chime in as well, I also love this project. I've been recommending it to all my coworkers. I use `--list` constantly. I just starting playing around with lua scripting and I'm excited to see the new updates on master.