tio/tio #324: Scripting and binary data

ii8, 2025-05-25:
Currently when reading input from the serial device in a Lua script, the input will get echoed to the console even though it has been read into a variable in Lua. This is not ideal when reading large amounts of binary data as all that will get vomited into the terminal.

Not echoing at all when a Lua script is consuming input would make sense to me, the script would then have the ability to selectively print all, some, or nothing of the input it reads, this is what I expected would happen. It could also be made configurable but that would slightly hurt portability of Lua scripts between configurations.

jcormier, 2025-06-12:
Hmm maybe it should be configurable.  If I was scripting something to interact with a unit, I would not expect the text to be consumed by the read calls. But I could see the value in having the option.  Just adding a different opinion.

lundmar, 2025-06-12:
> Currently when reading input from the serial device in a Lua script, the input will get echoed to the console even though it has been read into a variable in Lua. This is not ideal when reading large amounts of binary data as all that will get vomited into the terminal.

Sorry for the delayed response. I'm bogged down with work so got distracted away from this question. I'm fine with changing the behavior so it does not print while reading. Printing in scripts will be the responsibility of the user.

lundmar, 2025-06-12:
@ii8 I'm open to improve the lua support in the ways you have suggested but I simply don't have any time to do this myself currently. However, this is open source so I invite you to try make the improvements you suggested. All improvements are welcome.

ii8, 2025-06-14:
86f48a2fb689dad56d0e195b4735ef0d4661e8b2