tio/tio #330: Tio Lua module not found with built-in scripting | Warning: lua: [string "tio"]:1: attempt to index global 'tio' (a nil value)

jhammerberg, 2025-06-26:
# Tio Lua module not found with built-in scripting
- Version: `v3.9`
- Install/package: snap & Meson/manual build

## Description
Hello, I'm having an issue using the built-in tio Lua package for scripting. I can execute normal lua commands, like `print('test')` just fine, but as soon as I try to use the tio module, like `tio.sleep(5)`, it says that tio is a nil value (it doesn't exist).

I was able to replicate this on both the snap package and by building tio manually with meson (Lua interpreter and compiler already installed). It occurs with both the `--script` and `--script-file` options.

## Steps to Replicate
1. Install tio v3.9 (using snap or manual Meson build/install)
2. Make sure you have a serial port somewhere than tio can connect to.
(I used socat with this command: `socat -d -d pty,raw,echo=0 pty,raw,echo=0`)
3. Use either the `--script "tio.sleep(5)"` or `--script-file` pointing to a script with any use of a tio module. \
(The exact command I used is: `tio --script "tio.sleep(5)" /dev/pts/3`)
4. See the warning/error: `Warning: lua: [string "tio"]:1: attempt to index global 'tio' (a nil value)`

## Expected Behavior
Recognize the tio module, correctly execute the tio command, etc.

---

I believe I followed the correct process to execute a tio built-in command from Lua, unless I'm missing something the README.md doesn't say you have to import anything in your script for it to work, and trying to manually import tio with something like `local tio = require("tio")` doesn't work either.

Let me know if you need any more information, I'm missing something, or if this is a known problem, thank you!

jhammerberg, 2025-06-26:
Of course I immediately see what the issue is now that I've made the issue. It looks like the Lua API was overhauled about 2 weeks ago, with a matching updated README.md, but those changes haven't actually been pushed to a new release. So all I had to do was use the functions with the `tio.` at the beginning.

closed 2025-06-26 by jhammerberg

lundmar, 2025-06-26:
@jhammerberg I expect to make a new release soon which will be tio v4.0 - the main feature is the overhauled lua api.

AlbrechtL, 2025-07-29:
I had the same issue (snap version 3.9) and thanks to this issue report and realized that my `tio` version was too old. I complied from source - now it works!

@lundmar the script feature is a really great feature for embedded system development. Thanks for it!