tio/tio #332: Support COBS decoding

FerriteGiant, 2025-07-25:
It would be awesome to be able to decode a stream of COBS encoded data. And then extra cool to filter which packets are printed based on a user defined filter. For example, if the first byte matches a value or a range of values.

Being able to send a COBS encoded packet would be cool too, could be an optional variation when using "--input-mode line"

https://en.wikipedia.org/wiki/Consistent_Overhead_Byte_Stuffing

I'm sure you are busy but I figured it doesn't hurt to float the idea.

lundmar, 2025-07-26:
@FerriteGiant What kind of system do you connect to that speaks COBS?

FerriteGiant, 2025-07-26:
@lundmar In our org we use it all the time for framing streaming data between microcontrollers and application processors and/or PCs. We use a channelized framework/protocol for streaming different data streams over a single serial bus. Each packet has a header with a channel number and other info and is COBS encoded. This makes is trivial to know where each packet begins and ends. Even if there are data drops or corruption it's simple to get back on track.