tio/tio #334: Piping 'echo' output to tio ("non-interactive mode") leads to unconsistent results

ErlingSigurdson, 2025-08-11:
For quite a while I had been using tio in the non-interactive mode to control an ESP32 microcontroller using AT-style commands. I pipe the commands to tio like this: `echo "AT+LOADDIGITAL=TOGGLE" | tio /dev/ttyUSB0`

It used to work fine, but several months ago it ceased working: the microcontroller doesn't receive the commands nor does it send appropriate responses. At the same time I'm still able to successfully use tio in the interactive mode ('tio --input-mode line /dev/ttyUSB0') —  commands are sent OK, and I get valid responses.

What's even more strange, if I first use minicom in a similar fashion (`echo "AT+LOADDIGITAL=TOGGLE" | minicom -D /dev/ttyUSB0`), afterwards the non-interactive communication via tio becomes possible again, i. e. shell command  `echo "AT+LOADDIGITAL=TOGGLE" | tio /dev/ttyUSB0` can be used successfully. As if minicom sort of removes some obscured obstacle.

lundmar, 2025-08-11:
@ErlingSigurdson What version of tio are you testing on now? Do you know what version you used to use when it worked?

sur5r, 2025-08-11:
I have another idea based on the minicom observation: You could also check whether `stty -F /dev/ttyUSB0` returns different settings before and after running minicom.

ErlingSigurdson, 2025-08-11:
@lundmar, I've tried every version of tio starting from v.2.5, and the issue still persists. Taking into account that switching tio versions, a USB-UART chip and a PC (including an OS distro) didn't change anything, I hate to say that I now suspect my own code I've written for ESP32. But on the other hand, why does minicom still work for me when I pipe into it? And why does tio still work for me in the interactive mode? I'm utterly confused.

@sur5r, nice idea, I will post the results when I'll get them.

Vynce, 2025-08-25:
I'm seeing the same behavior with a CH340C on MacOS with tio 3.9. I captured the signal with a cheap logic analyzer (PulseView + [Cypress FX2](https://sigrok.org/wiki/Lcsoft_Mini_Board)). It seems like the baud rate isn't getting set correctly when piping input via stdin.

If I try to send an ASCII test string at 1200 baud:
```
echo -en "test\r\n" | tio -b 1200 /dev/cu.usbserial-2020
```
then it looks like the first byte gets corrupted and the remaining bytes are sent at 9600 baud:
[image]

If I run tio interactively and paste in the test string:
```
tio -b 1200 /dev/cu.usbserial-2020
```
then it works as expected:
[image]

Vynce, 2025-08-25:
In addition to minicom as a workaround, here's a oneliner using [uv](https://docs.astral.sh/uv/guides/scripts/) and [pyserial](https://pyserial.readthedocs.io/):
```
echo -en "test\r\n" | uv run --with pyserial python -c 'import sys,serial; s=serial.Serial(sys.argv[1],int(sys.argv[2])); s.write(sys.stdin.buffer.read()); s.close()' /dev/cu.usbserial-2020 1200
```
or a bit shorter if CLI args aren't needed:
```
uv run --with pyserial python -c 'import serial; s=serial.Serial("/dev/cu.usbserial-2020",1200); s.write(b"test\r\n"); s.close()'
```

ErlingSigurdson, 2025-10-26:
> (1) Make "restoring thread" wait until "writing thread" finishes all writing (most sleek).

This one I couldn't make work. Other options do work.

I've also tried simply adding a tiny pause at the beginning of `tty_restore` to give data slightly more time to move. It also helped.

@lundmar, hello. Can you give me a hint on which method of solving the issue you deem preferable? I can move in that direction then.